mavlink/qgroundcontrol #2903: UDP link to Mavros never connects on Linux

mhkabir, 2016-02-27:
@DonLakeFlyer @dogmaphobic

So this is an issue I've been fighting with over the last month. All testing was done with latest master builds.

I have Mavros on my MAV, proxying UDP to 2 hosts (QGC laptop, QGC tablet). Now the weird thing is that QGC on Android connects automatically, but on Linux (Ubuntu 14.04) it never connects. I've tried a load of different stuff like opening ports, trying a different one, adding a link manually on a different port, etc. but nothing works. I've tried sending to different ports from the MAV side, and I've also tried MAVProxy, but it seems that nothing can get a UDP connection into QGC.

Another observation - If I have something doing a local UDP broadcast (say PX4 SITL or another mavros/mavproxy instance connected  to a Pixhawk via USB), QGC connects fine. I never made any configuration changes on my Ubuntu system that would cause it to break as far as I understand. IIRC, it stopped connecting ~ 2 months ago. 

Some help would be greatly appreciated!

DonLakeFlyer, 2016-02-27:
I just did this:
* Cranked up arduplane SITL running a continuous mission. This talks UDP.
* Built and run latest master QCG

Everything auto-connected fine. This looks like be either an OS or Mavros problem. QGC seems to be working fine on UDP links.

DonLakeFlyer, 2016-02-28:
I'm going to close this. Unless you have more direct information I don't see how this is a QGC issue.

closed 2016-02-28 by DonLakeFlyer