mavlink/qgroundcontrol #2910: can't connect to Pixhawk via Bluetooth on Android

kd0aij, 2016-02-28:
"service not available"

DonLakeFlyer, 2016-03-03:
@dogmaphobic Should bluetooth work on Android?

dogmaphobic, 2016-03-30:
What device? The TBS Crossfire works. At least it was working the last time I used it.

gregswf, 2016-05-09:
I'm using bluetooth with PX4 stack, it works fine on ASUS zenpad 7 but it does not connect on Galaxy Note4, I get "service not available"

dogmaphobic, 2016-06-07:
I need to know what "Bluetooth" device is plugged to the Pixhawk.

kd0aij, 2016-06-07:
Mine shows up as HC-06 and the chip is labeled BT417

dogmaphobic, 2016-06-07:
This? I need to get one to test. The only Bluetooth device I have is the Crossfire.

[image]

dogmaphobic, 2016-06-07:
I found a datasheet for it. Did you have to set it up using the "AT" commands as described?

https://www.olimex.com/Products/Components/RF/BLUETOOTH-SERIAL-HC-06/resources/hc06.pdf

kd0aij, 2016-06-07:
It's been 2.5 years since I got mine. Don't remember, but I probably did set the baud rate using an AT command first.

kd0aij, 2016-06-07:
And that one looks almost identical to mine: http://smile.amazon.com/Kootek%C2%AE-Bluetooth-Serial-Module-Arduino/dp/B00DQ4A7O6?ie=UTF8&psc=1&redirect=true&ref_=oh_aui_search_detailpage

dogmaphobic, 2016-06-08:
Got it in the mail today.

The good news, I can't get it to connect either. Meaning I could reproduce the issue and I can debug it. The bad news, I won't be able to do much for the next couple of weeks 8P

DonLakeFlyer, 2016-06-08:
@dogmaphobic Do you think this is important enought to hold 3.0 release for?

dogmaphobic, 2016-06-08:
Good question. Bluetooth was ever really tested using only the Crossfire. Today was the first time I connected something else. It took me an hour just to get the thing configured and paired with a Nexus 9. Bluetooth is inherently unreliable. 

I don't think it's enough to hold 3.0 as I don't think that many people use it to begin with. Those who try, it will either work or not, which is quite *standard* with Bluetooth things, specially a $5 trinket from China.

kd0aij, 2016-07-08:
@dogmaphobic update: can't connect via wifi using Moto G running Android 6.0, but Nexus 7 running Android 6.0.1 works fine

kd0aij, 2016-08-04:
as of today, my Moto G won't connect via Bluetooth either. Any interest in why something that used to work is now broken? Or is BT deemed to be of no value for this application?

kd0aij, 2016-08-04:
update: the free "BlueTerm" app connects to the BT dongle and displays data on both Nexus7 and MotoG

DonLakeFlyer, 2016-08-06:
@dogmaphobic Did you ever get a bluetooth thingy? I don't have anyting to test with.

hsteinhaus, 2016-08-11:
I have a similar problem with a HC06 bluetooth module on Android here  (Beanstalk 6.0, Android 6.0.1 on Amazon Tate), see attached log. 

mtp://[usb:003,087]/Internal%20storage/Download/log.txt

I particularly wonder about the message "Using found rfcomm channel 0". When I use the HC06 under plain Linux, I have to bind to channel 1 to see the MAVLink data stream from the HC06 .

baucho, 2016-09-10:
Same here i think!
My galaxy tab s2 connects to bluetooth but noch data Coming 
[image]
[image]
[image]
[image]
[image]

DonLakeFlyer, 2016-11-11:
@dogmaphobic Would be nice to figure this one out for 3.1 release. Lots of request for it.

DonLakeFlyer, 2016-12-04:
Working now.

GlidLov, 2017-01-10:
same here, service not avaiable. Looking forward for it

DonLakeFlyer, 2017-01-10:
@GlidLov Are you using the latest version of QGC?

GlidLov, 2017-01-12:
@DonLakeFlyer yes

DonLakeFlyer, 2017-01-12:
That looks like the same board I tested with. I'll try it again today.

Patnetpatnet, 2017-02-06:
I'm still not able to connect to BT via Android even with the new version. Just crashes the QGroundcontrol app on attempted connection to BT.

Siytek, 2017-02-16:
I also have the same problem as Patnetpatnet, QGC just crashes when I try to open a BT connection. I set up connections to both my HC-06 and to my TBS Crossfire, both with the same issue. Running on a brand new Acer Iconia, which I bought today for dedicated QGC usage, running Marshmallow 6.0.

DonLakeFlyer, 2017-02-16:
@dogmaphobic Do you time to look at this?

Siytek, 2017-02-17:
Thanks for a quick response on re-opening the issue. Let me know if there is anything I can do to assist. Would it be worth me having a go at putting Qt on my Android device, compiling the source and seeing if I can feed back an error report?

dogmaphobic, 2017-02-17:
I found the change that is causing the crash: #4349 

It changed how the configuration is handled. The Bluetooth driver is now using a NULL pointer for its configuration and crashing. I need to understand what the change was for before I can find a fix.

Siytek, 2017-02-18:
I just compiled QGC in Qt and deployed to my tablet. I checked that the addition from #4565 was present before compiling, the following line was present in BluetoothLink.cc

    , _config(qobject_cast<BluetoothConfiguration*>(config.data()))

Still crashing on opening the BT connection. Application output errors as follows

```
D QGroundControl: (null):0 ((null)): LinkManager mavlinkStatus:channel:flags 0xe12ec9d0 2 2
F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x18 in tid 30176 (QtThread)

"org.mavlink.qgroundcontrol" died.
```

Apologies if I'm doing something dumb here, I'm new to Qt! :)

dogmaphobic, 2017-02-18:
> I just compiled QGC in Qt and deployed to my tablet.

If you are building yourself, you can launch it through the debugger and see where it crashes.

DonLakeFlyer, 2017-02-18:
@Siytek If you're new to Qt and you figured out how to get an android build to build and deploy then that's huge! That combination can be quite the pain. Anyway, if you could get the crash in the debugger and find the crash point that would be awesome.

Siytek, 2017-02-19:
Thanks @DonLakeFlyer, I'm quite well rehearsed with lower level stuff, I'm a hardware guy really (electronic engineer as a day job) but would like to improve my programming/embedded skills. There's some pretty good documentation so didn't have too much problem deploying to Android.

Sure, I was just in the process of reading up on the debugger but had to go out and do some socialising. I'll continue tomorrow and try find the point in which the crash occurs.

Siytek, 2017-02-19:
@DonLakeFlyer @dogmaphobic I've been scratching my head for a few hours now, when I deploy the debug build, it successfully runs on the target but debugging fails with the error

Debugging starts
Failed to contact debugging port.Debugging has finished

Debugging works ok with the Qt example project, I can set breakpoints etc. Did I miss something?

DonLakeFlyer, 2017-02-19:
Have you turned on developer mode on your device?

Siytek, 2017-02-19:
Yes USB debugging is active on my device, it works fine with one of the Qt sample applications. I am probably getting a bit off topic here with Qt related issues, ill head over to QGC in Google groups to start a new topic.

DonLakeFlyer, 2017-02-19:
> it works fine with one of the Qt sample applications

Yeah, sorry I missed that. Maybe Gus will have some ideas.

DonLakeFlyer, 2017-02-19:
I can repro the crash, but I can't catch it in the debugger either.

DonLakeFlyer, 2017-02-19:
Actually I can't repro the crash. Here's why:
* I didn't uninstall the previous version of QGC I had on the device
* Then when you try to deply a new build from Qt it is not super clear that in reality it didn't deploy a new one. There is a failure message is you look for it.
* Instead it cranks up the old build that was already on there which is still bad.
* And you can't debug it, because it isn't what you built.

Once I deleted QGC completely from the device and then really deployed a new version from QtCreator using master it all worked.

Also just discovered that daily builds are not getting pushed up correctly. They are a week old. So daily builds don't have the fix.

Siytek, 2017-02-20:
@DonLakeFlyer you are correct! Makes perfect sense, its always something simple :) I deleted the original installation, debugging now working for me and BT didn't crash - I didn't assign an adapter nor try a connection but I now get the disconnect button available after pressing connect. Ill try it on the plane tomorrow.

DonLakeFlyer, 2017-02-20:
Great. Let us know if everything is working for you now and we can close this one out.

Patnetpatnet, 2017-02-20:
Does the latest QGC in Google Play now have a BT Android fix? Or does one have to go this circuituitous route? Not clear. BT issues have been outstanding for a long long time btw. Thanks,

DonLakeFlyer, 2017-02-20:
@dagar I just went to play store dev console and it says there is no APK in beta testing. Am I looking at the wrong place, or is something out of whack. Want to make sure Android beta is up to date with latest daily.

Siytek, 2017-02-20:
I just deployed a release build on my tab and I am pleased to report Bluetooth is now working as expected! At a glance, all instruments and information now seems to be displaying as expected in QGC. Thanks for resolving this issue, I'm excited to go try my tablet in the field now!

DonLakeFlyer, 2017-02-20:
Thanks for testing folks.

FYI: Still trying to figure out Android daily builds from Google Play.

Siytek, 2017-02-24:
Did you manage to get daily builds sorted? @Patnetpatnet if you really want a copy with working Bluetooth and it's not going to be available on Google Play just yet, I can have a go at sending you a compiled version of the latest code.

dagar, 2017-02-24:
The Android Google Play beta (daily build) isn't currently auto updating, but you can install it manually.
https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.apk

Patnetpatnet, 2017-02-24:
Thanks! Will test this weekend.

hackra76, 2017-02-24:
The version of Google play beta is 3.1.3, but the version posted here is 3.1.2-277.... Couldn't this be the problem why it is not updating ?

DonLakeFlyer, 2017-02-25:
> Did you manage to get daily builds sorted?

As far as I know google play still does not have the right daily build. @dagar ?

dagar, 2017-02-25:
Yes, that's the issue with the beta/daily build, and it won't auto update without a version greater than the stable. There will be a separate play store entry for the QGroundControl beta.

Patnetpatnet, 2017-02-26:
Was able to successfully test that the beta via the link above from Daniel does not crash the app as the final version does. However, I'm back at square one in that I cannot maintain a connection. It keeps cutting out. Something likely to do with the fact I just use BT as a bridge to an Xbee link to Mavlink Pixhawk. Historic problem going back years for me.

Siytek, 2017-02-26:
@Patnetpatnet I had problems just like you describe with MAVLink when just using OpenLRS serial bridge for telemetry. A serial pass through connection will not manage any lost MAVLink heartbeats, therefore when a heatbeat packet is dropped, QGC will think that the connection has been lost, leading to sporadic disconnection warnings. 

I now use TBS Crossfire which has a dedicated setting for MAVLink that overcomes these problems as the radio firmware manages the MAVLink protocol as well as passing the serial data back and forth. I have never used Xbee so I'm not 100% sure, but it sounds like an issue with your setup, not a software issue with QGC. Maybe this will help? https://pixhawk.ethz.ch/tutorials/how_to_configure_xbee

Patnetpatnet, 2017-02-26:
Thanks Simon. The only change I could make would be dropping the now configured 115200 baud rate down to the recommended 57600. It works two-way as is in Tower though. Wondering whether that's the issue?

Siytek, 2017-02-26:
Its difficult to say without knowing exactly what setup you have but yes you could try reducing the baud rate to minimise packet loss, also make sure all baud rates at matched up throughout your telemetry connection. 

This is probably off topic now for this thread as it is unlikely to be related to QGC BT issues, your best bet would be to open a new topic here (assuming you're using PX4 flight stack) https://groups.google.com/forum/#!forum/px4users
and provide as much detail as you can about your hardware and software setup.

Patnetpatnet, 2017-02-28:
No longer Android-specific for sure. Same connection loss on Windows via BT and Xbee.

DonLakeFlyer, 2017-03-01:
@Siytek You aren't getting communication list problems with your setup right?

Patnetpatnet, 2017-03-01:
Correct. Just intermittent connection lost. Always when sending data.

Siytek, 2017-03-01:
@DonLakeFlyer no my setup works perfectly now as far as my bench testing goes, ill fly it soon though as the weather is starting to pick up. Im confident it will work great now but ill report back if I have any issues.

I used to get random disconnection warnings due to lost heartbeat packets, but this was because my old telemetry radio setup did not have flow control, not because of a bug with QGC/PX4. I believe this is the problem @Patnetpatnet currently has with his setup. Its an issue with the radio/telemetry uplink, not a fault of the software.

DonLakeFlyer, 2017-03-02:
@Siytek I'm going to leave this issue open to you report back on flying. If that goes well then I'm going to close this issue as resolved.

[2 more replies]

closed 2017-03-02 by DonLakeFlyer