mavlink/qgroundcontrol #2920: qgc on Mac, "vehicle not responding" and crashes

rcorbari, 2016-03-02:
Hello,

I cannot connect my 3DR Pixhawk to qgc on a Mac, latest app official release.

The connection via USB fails and the message is "your vehicle is not responding. If this continues please check that you have an SD Card inserted and try again."

The SD card is in, I also tried to change it, and I also tried both ExFAT and MacOS formatting.

Furthermore, when I close the app, it crashes.

Anyone has the solution to this issue?

APM Mission Planner works flawlessly.

DonLakeFlyer, 2016-03-02:
I assume since you mention AP2 and MP you are running ArduPilot. That message is not quite telling you the right thing, we need to change it. It is more likely that parameters are not coming across the connection. Can you try this:
* Don't boot QGC
* Plug the board in via USB, wait for it to fully boot (white light)
* Now start QGC

Also can you tell me what version of OSX you are using? El Capitan has crazy USB problems. Also what version of ArduPilot?

rcorbari, 2016-03-02:
Thanks Don for your prompt support.

I tried to flash both corresponding firmwares on APM Planner and QGC. The last one is the QGC FW which has been flashed after a couple of QGC crashes...

I'm using MacOS 10.11.3 (El Capitan) with an iMac 27 late 2013.

At the moment QGC doesn't start at all. I also tried the daily build and the second last one. Let me reboot and I'll be back here.

DonLakeFlyer, 2016-03-02:
Ooh, El Capitan. That's going to be rough. I couldn't get AP2 to not be flaky connecting on El Capitan either. It tends to just hang. Does AP2 connect consistently well for you on El Capitan?

rcorbari, 2016-03-02:
AP2 went well so far. Do you mean that I need to downgrade if I want to use QGC?

Isn't possible to solve these issues on El Capitan? Asking to downgrade to all of us it's not a good thing :)

rcorbari, 2016-03-02:
BTW, my Pixhawk main LED light never worked. Do I have to activate it somewhere?

Only the smaller LEDs work.

DonLakeFlyer, 2016-03-02:
I'll try AP2 again. Have you just tried using AP2 recently or have you been using it succesly on El Captian for a number of months. If they can run we should be able to figure something out. 

If you search for El Capitan USB problems or El Capitan usb Arduino you'll see all the issues. I've reported the problems to Apple, as have other people but have heard nothing back.

DonLakeFlyer, 2016-03-02:
> my Pixhawk main LED light never worked. Do I have to activate it somewhere?

Not that I know of. Sounds like a hardware problem.

rcorbari, 2016-03-02:
Just tried recently, since this is my first Pixhawk programming.

DonLakeFlyer, 2016-03-02:
Ok. I'm checking with the AP2 developer to see what he is hearing on El Capitan problems.

rcorbari, 2016-03-02:
Ok thanks. AP2 was ok so far, but now it can just write FW to Pixhawk. Like QGC, even if QGC also crashes.

I'm trying a different cable now. I already tried different USB ports.

rcorbari, 2016-03-02:
About the LEDs, the remote safe button (the one sold with the Pixhawk) keeps on flashing fast. About the LEDs on the Pixhawk, FMU has just the blue power LED on (no B/E), while IO LEDs are power solid green, B/E solid orange and ACT flashing blue.

DonLakeFlyer, 2016-03-02:
I believe the safety button flashes when it is still in safety. Hold it and it will go solid which means you can arm. The other littles one I don't know.

rcorbari, 2016-03-02:
Holding the safety button doesn't change its state. Still 4hz blinking.

DonLakeFlyer, 2016-03-02:
It's likely angry at you because there is still something you need to calibrate

rcorbari, 2016-03-02:
Ok yes it's possible because I re-flashed the AP2 FW.

rcorbari, 2016-03-02:
Got the thing about the main LED. Is it possible it doesn't light because I'm using its external LED?

rcorbari, 2016-03-02:
Is there a way to blank format the Pixhawk and re-flash the FW? It seems that I'm stuck with the new AP2 FW and I cannot connect to the Flight Controller. It just connects for flashing.

QGC is worse because it crashes like crazy.

Strange because until yesterday at least AP2 worked.

DonLakeFlyer, 2016-03-02:
Have you installed the righ FTDI drivers as specified here: http://ardupilot.com/forum/viewtopic.php?f=82&t=14646&p=41679#p40524? That may be the problem with QGC

rcorbari, 2016-03-03:
Step by step, I erased the FTDI drivers from Terminal, rebooted and flashed the Pixhawk FW from QGC which I installed fresh from download (Mac preferences and cache erased). FW flashing is successful, then... nothing. QGC doesn't recognise the Pixhawk, if I click on the Firmware tab again, the app crashes.

I will install the FTDI drivers again tomorrow, I'm in Italy and it's late now :)

Thank you very very much for your support, get in touch tomorrow.
[image]

DonLakeFlyer, 2016-03-03:
Awesome, thanks for working through it.

rcorbari, 2016-03-03:
Hello.

No way. I tried to remove then reinstall the FTDI drivers, but I can only flash the FW (successfully) but then it's impossible to connect again via USB.

Always the same error: ""your vehicle is not responding. If this continues please check that you have an SD Card inserted and try again."

This is really a major issue.

DonLakeFlyer, 2016-03-03:
Let's take it step by step to try to remove any variables:
* Your sd card should be formatted FAT32
* Install the latest OSX daily build of QGC from here: http://qgroundcontrol.s3.amazonaws.com/master/qgroundcontrol.dmg?dl=0
* Boot QGC (don't have your board plugged in)
* Go to Firmware Upgrade
* Plug in you board
* Select PX4 Stack
* Flash the board
* When Flash is complete, unplug board, shutdown QGC
* Plug board back in, board should boot and after boot is finished big LED should go red
* Now start QGC

QGC should connect

DonLakeFlyer, 2016-03-03:
You may get the SD card message, but if you should be able to dismiss that and it will still connect.

rcorbari, 2016-03-03:
Hello @DonLakeFlyer 

Meanwhile, I really tried everything, including editing the Info.plist file in /Connect folder for USB issues.

At the moment I tried to flash another vehicle (plane, then rover) without success (both on WGC and APM). Next step has been power the Pixhawk holding down the safety button, and thanks lord something happened. QGC found the FC, but re-flashing its (QGC) firmware, QGC app crashed. On my system it's very very unstable. Now I'm trying to finish this FW flashing, then I'll try your steps.

I'm in front of my Mac working right now.

P.S.: I also tried with another Macbook, no success.

Thanks.

rcorbari, 2016-03-03:
Actually I'm already following the steps right now...

rcorbari, 2016-03-03:
Crash of the QGC app flashing the firmware...

`Process:               qgroundcontrol [1606]
Path:                  /Applications/qgroundcontrol.app/Contents/MacOS/qgroundcontrol
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               2.0.0 (227)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           qgroundcontrol [1606]
User ID:               501

Date/Time:             2016-03-03 17:32:28.186 +0100
OS Version:            Mac OS X 10.11.3 (15D21)
Report Version:        11
Anonymous UUID:        26BE420E-F5F3-3070-83AE-2703925761C4

Time Awake Since Boot: 4800 seconds

System Integrity Protection: disabled

Crashed Thread:        11  QThread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x8:
--> 
    __TEXT                 000000010e719000-000000010edcf000 [ 6872K] r-x/rwx SM=COW  /Applications/qgroundcontrol.app/Contents/MacOS/qgroundcontrol

Thread 0:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtCore         	0x000000010f095270 0x10f073000 + 139888
1   org.qt-project.QtCore         	0x000000010f0953c9 QInternal::activateCallbacks(QInternal::Callback, void**) + 25
2   org.qt-project.QtCore         	0x000000010f2551e2 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 1010
3   libqcocoa.dylib               	0x0000000113932b0e 0x113911000 + 137998
4   libqcocoa.dylib               	0x0000000113933391 0x113911000 + 140177
5   com.apple.CoreFoundation      	0x00007fff937f55c1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
6   com.apple.CoreFoundation      	0x00007fff937e741c __CFRunLoopDoSources0 + 556
7   com.apple.CoreFoundation      	0x00007fff937e693f __CFRunLoopRun + 927
8   com.apple.CoreFoundation      	0x00007fff937e6338 CFRunLoopRunSpecific + 296
9   com.apple.HIToolbox           	0x00007fff9ab93935 RunCurrentEventLoopInMode + 235
10  com.apple.HIToolbox           	0x00007fff9ab9376f ReceiveNextEventCommon + 432
11  com.apple.HIToolbox           	0x00007fff9ab935af _BlockUntilNextEventMatchingListInModeWithFilter + 71
12  com.apple.AppKit              	0x00007fff9119e0ee _DPSNextEvent + 1067
13  com.apple.AppKit              	0x00007fff9156a943 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
14  com.apple.AppKit              	0x00007fff91193fc8 -[NSApplication run] + 682
15  libqcocoa.dylib               	0x00000001139322df 0x113911000 + 135903
16  org.qt-project.QtCore         	0x000000010f251c1c QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 412
17  org.qt-project.QtCore         	0x000000010f254a95 QCoreApplication::exec() + 341
18  org.qgroundcontrol.qgroundcontrol	0x000000010e7f293c 0x10e719000 + 891196
19  org.qgroundcontrol.qgroundcontrol	0x000000010e7224a4 start + 52

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff94f9d6de __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff999cf729 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff999cd365 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff94f9dff6 kevent_qos + 10
1   libdispatch.dylib             	0x00007fff94eaa099 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff94ea9d01 _dispatch_mgr_thread + 52

Thread 3:: QThread
0   libsystem_kernel.dylib        	0x00007fff94f9d176 __select + 10
1   org.qt-project.QtCore         	0x000000010f2a27a3 qt_safe_select(int, fd_set*, fd_set*, fd_set*, timespec const*) + 547
2   org.qt-project.QtCore         	0x000000010f2a36b5 QEventDispatcherUNIXPrivate::doSelect(QFlags<QEventLoop::ProcessEventsFlag>, timespec*) + 693
3   org.qt-project.QtCore         	0x000000010f2a488a QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 234
4   org.qt-project.QtCore         	0x000000010f251c1c QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 412
5   org.qt-project.QtCore         	0x000000010f0a707e QThread::exec() + 110
6   org.qt-project.QtCore         	0x000000010f0aac02 0x10f073000 + 228354
[... 778 lines cut ...]
JS JIT generated code             1172K       34 
JS VM register file               8192K        4 
JS garbage collector              3584K       12 
Kernel Alloc Once                    8K        3 
MALLOC                           163.5M       55 
MALLOC guard page                   32K        7 
Memory Tag 242                      12K        2 
OpenCL                               8K        2 
OpenGL GLSL                        256K        4 
Process Corpse Info               2048K        2 
STACK GUARD                       56.2M       43 
Stack                             28.4M       43 
VM_ALLOCATE                        808K       45 
__DATA                            30.1M      264 
__GLSLBUILTINS                    2588K        2 
__IMAGE                            528K        2 
__LINKEDIT                        96.2M       53 
__TEXT                           182.8M      264 
__UNICODE                          552K        2 
mapped file                        1.2G       32 
shared memory                     16.4M        9 
===========                     =======  ======= 
TOTAL                              1.8G      919 

`

rcorbari, 2016-03-03:
First comments:

I repeated your steps 3 times. Once QGC crashed during flashing, 2 time it crashed quitting the app, third went OK.

The Pixhawk LED light does NOT go fixed red, but is fast flashing red (probably 4hz).

The SD card has been reformatted using disk utility to FAT (GUID).

QGC doesn't see the FC.

The only little success I had is when I flashed the APM plane from Mission Planner and - then - tried the reset holding down the safety button and powering the Pixhawk. The FC made some different sounds then the safety button's light went blinking red. When I connected the USB cable, FC was visible.

DonLakeFlyer, 2016-03-04:
I don't think holding the safety button while powering on does anything special. If it does, I wouldn't do that since I don't know what that might do. Flashing red is an ok state, just means you need to calibrate sensors.

It is very odd that you are getting so many crashes. We literally have gotten no reports of crashes on OSX builds.  The stack unfortunately isn't very helpful as a hint to the problem.

Just to check are you saying to using Mission Planner and/or APM Planner 2 everything is working fine. Both flashing and connecting?

DonLakeFlyer, 2016-03-04:
I think I found the crash when you are flashing ArduPilot firmware.

DonLakeFlyer, 2016-03-04:
I believe I found the shutdown crash as well. Both the flash and shutdown are new problems in daily builds introduced in the last couple weeks. There were so new I just hadn't been hitting them yet.

rcorbari, 2016-03-05:
Hi @DonLakeFlyer 

With APM Planner 2 I can both flash and setup or control. Didn't try Mission Planner since I think it's for Windows OS only.

About the crashes, I had them also before to install the daily build.

DonLakeFlyer, 2016-03-05:
The daily build should have the flash and shutdown fixes in it. Hopefully I fixed the ones you are hitting.

rcorbari, 2016-03-06:
Thanks @DonLakeFlyer I will try it ASAP.

closed 2017-02-18 by DonLakeFlyer