mavlink/qgroundcontrol #2922: Planning map screen

lvale, 2016-03-02:
I would suggest to make it easier to plan and follow a mission that either a small text label would be placed near the map point. If taken to the latest consequences we could deprecate the right side mission items bar and clicking on the mission item on the map would open the "configuration" for the point.

[image]

DonLakeFlyer, 2016-03-04:
Maybe something along these lines fro 3.0

tubeme, 2016-04-11:
Very nice idea i must say. The current implementation does not work well for missions with more than 10-15 points depending on the screen size. If you have a mission with 50 points as I did it it nightmare, scrolling down the waypoint list, the same time scrolling the map screen... BTW I did try to edit a mission with 30 points in the OS X version... with just the mouse pad of the macbook and it was a disaster!!! 

If this suggestion is accepted we will no longer need to have the waypoint list, giving us more space for mission planning concentrating on landscape and mission objectives.

There is another problem editing the mission. When I change the altitude of a point it does not apply the same alt to the next point. So I have to manually change the altitude of most of the waypoints. Sometimes it remembers my last alt... I dont understand what is the principal...

We would love to have a kind of additional overlay view, like in the game SimCity, that has the same mission spots but somehow indicates the altitude of the mission and its point to point distance and cumulative distance beside every waypoint. This will give the operator a 3D image of the flight. If lets say by coloring from green to red the waypoints on the map and the links between them we can indicate the current altitude of the planned flight. 

What do you think?

DonLakeFlyer, 2016-04-11:
> When I change the altitude of a point it does not apply the same alt to the next point. So I have to manually change the altitude of most of the waypoints. Sometimes it remembers my last alt... I dont understand what is the principal...

Can you add a separate issue for this? It should be working.

DonLakeFlyer, 2016-04-11:
> If you have a mission with 50 points as I did it it nightmare, scrolling down the waypoint list, the same time scrolling the map screen

This should have been fixed as well. I'll put in issue for both these things (and above one).

tubeme, 2016-04-12:
@DonLakeFlyer It is true that when I click on a point it selects it, but from point # 10 on the selected waypoints stay at the bottom of the screen, so in most cases you don't see the values, so you have to mandatory scroll down in order to change the numbers. So again it is in reality selected but not usable this way. If I can use the right click over point to see the waypoint and change its parameters I think would be more practical.

If we need to change a bulk of data on the waypoints it is not a bad idea to have a Table view from the mission, Excel style. Thus making bulk changes a lot easier. This view can be be activated with a toggle view button or something. Also it must show all the attributes of a waypoint on one line of the table, ie real altitude, rel, altitude, coordinates, speed, cumulative distance of the mission, relative distance from previous point etc.

It is not a bad idea to have the option for naming the waypoints!!! If we have to do everiday missions with a certain routine, and then putting this process in a workflow, most likely the missions will be created by one person, authorized by another and executed by a field guy. Then we mandatory need some sort of this Excel style view and naming some waypoints and POIs in order to have a workflow that includes a role separation and networking process.capabilities.

DonLakeFlyer, 2016-04-13:
> It is true that when I click on a point it selects it, but from point # 10 on the selected waypoints stay at the bottom of the screen, so in most cases you don't see the values, and you have to mandatory scroll down in order to change the numbers.

This is fixed now.

DonLakeFlyer, 2016-05-13:
@NaterGator Has made some other fixes here to the scrolling I think. I think it's good for 3.0.

NaterGator, 2016-05-13:
:neutral_face: That sounds vaguely familiar but I don't specifically remember what I fixed here. 

My respective suggestion for the original topic would be to use accents to the stroke/border of the point marker on the map to indicate its functionality as opposed to cluttering the map with a bunch of glyphs.

For example something like this for loiter points:
[image]

And maybe crosshairs for navigation points, etc. That gives the user a very quick assessment of what they can expect the craft to do at that point without taking up unnecessary adjacent space (or creating oddly shaped/sized waypoint indicators).

DonLakeFlyer, 2017-03-25:
Most of this is done now. If there is still things that pop up we can create new issues. This one has gotten too cluttered with things already fixed.

If you have things in here that you don't see in tomorrows daily build. Please enter new seperate issues for each.

closed 2017-03-25 by DonLakeFlyer