mavlink/qgroundcontrol #2926: Restart of QGC required to refresh magnetometer calibration parameters

kd0aij, 2016-03-03:
1) added an external mag to a quadcopter which had been calibrated and flown previously
2) booted, preflight check failed "mag0 uncalibrated" (hopefully this message is less confusing in latest master)
3) ran compass calibration in QGC
4) no mag1 cal parameters displayed by QGC
5) reboot pixhawk and refresh parameters in QGC; still no mag1 cal params displayed; pixhawk console shows they are assigned properly
6) restart QGC: parameters now displayed correctly

If the mag parameters aren't refreshed properly, perhaps this indicates a more general problem with QGC parameter handling.

DonLakeFlyer, 2016-03-03:
I just noticed this as well today. I wonder if it is more fallout from calibration message string changes on the firmware side. Not sure. That's the only thing I can think of, since QGC mag cal code hasn't changes in quite a while.

DonLakeFlyer, 2016-03-03:
Or maybe param cache? Not sure. Odd stuff.

DonLakeFlyer, 2016-03-06:
@LorenzMeier This is a firmware optional parameter issue.

closed 2016-03-11 by DonLakeFlyer