mavlink/qgroundcontrol #2927: Retrieving of SRTM data for Terrain following

lvale, 2016-03-04:
@DonLakeFlyer I was looking if QGC was retrieving SRTM data for the Terrain following on ArduPilot, and didn't find If not implemented yet,do you have a time-frame ?

DonLakeFlyer, 2016-03-04:
Nope. No timeframe for that yet. Not in next 3.0 release.

ebennb, 2017-01-08:
Any updates here? This is a critical feature for me as I do surveys' in areas with much topography. Thanks!

DonLakeFlyer, 2017-01-08:
Scheduled for next release. No timeframe yet.

dagar, 2017-01-08:
I was thinking about this a few months ago. Would you want to feed it to the autopilot over mavlink or use it at mission planning time?
There also seems to be different sources of the data and they're not necessarily easy to pull from. We might need to host a version of it? http://www.cgiar-csi.org/data/srtm-90m-digital-elevation-database-v4-1

lvale, 2017-01-09:
@dagar 90m resolution is not very helpful. I can't remember where I've seen 30m data, but it exists...

dagar, 2017-01-09:
There's this one http://www.cgiar-csi.org/1106, but the site isn't loading.

dagar, 2017-01-09:
Good data is part of it, any thoughts on the implementation @lvale? Does QGC cache SRTM and feed it to the vehicle over mavlink, or could we use it during planning with absolute altitude waypoints?

birchera, 2017-01-09:
There is the [Google Elevation API](https://developers.google.com/maps/documentation/elevation/intro). It has a few meters accuracy, which would be decent for terrain following I guess.

dagar, 2017-01-09:
That might be a good option. The last time I looked I only found the javascript client library, but I see they have server side options now. There are pretty low API limits, how do we handle that? Allow the user to bring their own key?

kikislater, 2017-01-09:
Better option is not google IMO, it's SRTM 30. Google is not proven as a good provider in GIS ...
I have made a request here : https://github.com/mavlink/qgroundcontrol/issues/4007

birchera, 2017-01-09:
What are the reasons you doubt the google data? They provide accuracy of their data for every single point and the number is usually below 5m from what I have seen.

On the other hand, 30m accuracy is not very helpful if you plan surveys on e.g. 45m altitude.

kikislater, 2017-01-09:
Google use SRTM data's and his own algorithm to provide data's on elevation. Algorithm is interpolation and could provide accurate data when you divide the number by 3. So an SRTM30 you could be around 10m but I prefer raw data's than interpolated data's by Google.
Google is not a provider of elevation data but use data from other provider. Sometimes they made mix from aster dem which is definitively less and none accurate than an SRTM product ...

dagar, 2017-02-03:
https://github.com/Skycatch/srtm-downloader

kikislater, 2017-02-03:
Interesting, I didn't know go language before but only srtm 3 which mean 3 arc second or 90 meter resolution. 
What I purpose above is srtm 1 data's (1 arc second = 30 meter resolution). 

So you are not interested with a python purpose. Which language you need ? What is the goal of these go language inside qgroundcontrol (it's not to criticise, only for my information)? 

Terrain following with ardupilot have been tested in usa with srtm 1 data's but mission planner with gmap.net provide only srtm1 data only for united states. 

 if you watch Mission planner source code :

[url]https://github.com/ArduPilot/MissionPlanner/blob/master/ExtLibs/Utilities/srtm.cs[/url]

You could watch it download tiles from here :
1) For US : [url]http://firmware.eu.ardupilot.org/SRTM/USGS/SRTM1/version2_1/SRTM1/[/url]
2) And here for the rest of the world : [url]http://firmware.eu.ardupilot.org/SRTM/[/url]

There is a region definition in 1) (a map) which refer only to SRTM30 in US.

I downloaded a tile from the rest of the world in Europe (France) and opened in GIS. You get only 90m resolution

dagar, 2017-02-03:
Not the code itself, but as yet another example of one a data source and how it's structured.

lvale, 2017-03-25:
@DonLakeFlyer Given up or delaying ?

DonLakeFlyer, 2017-03-25:
Not giving up. Just hasn't popped to the top of the list yet.

closed 2026-10-04 by github-actions[bot]