mavlink/qgroundcontrol #2929: The takeoff waypoint should not be clamped to home

AndreasAntener, 2016-03-04:
Noticed in current daily build, wasn't like this before.

- I think generally it doesn't make sense
- When I now load a mission from file and have no vehicle connection it just positions home in the center of the mission, and not anymore where it initially was.

AndreasAntener, 2016-03-04:
Eh, I just noticed this is only the case if I'm not connected to a vehicle

DonLakeFlyer, 2016-03-04:
If you are not connected to a vehicle, there is a setting which specifies which firmware you are designing missions to. By default it is set to ArduPilot. The takeoff command in ArduPIlot does not support a position. If you switch the setting to PX4 (or connect to a PX4 Vehicle), you will be able to set a position on takeoff.

closed 2016-03-04 by DonLakeFlyer