mavlink/qgroundcontrol #2938: Figure out iOS build for travis

DonLakeFlyer, 2016-03-06:


jcooper21, 2016-03-07:
Is QGC going to be available on iPad minis? That would be awesome

DonLakeFlyer, 2016-03-07:
Yes. In fact if you have the ability to build QGC yourself, you can run there right now.

jcooper21, 2016-03-10:
That is great except I definitely to not have the skills to do that.

DonLakeFlyer, 2016-03-13:
```
        # Install handcrafted Qt ios version
        wget https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/Qt5.5.1-ios.tar.bz2
        mkdir -p ~/qt-ios
        tar jxf Qt5.5.1-ios.tar.bz2 -C ~/qt-ios
        export PATH=~/qt-ios/ios/bin:$PATH
```

```

jcooper21, 2016-03-16:
Is there any chance of the iOS builds showing up in the apple flight test app for beta software?  If there is I am happy to test it for you.

DonLakeFlyer, 2016-03-24:
https://www.objc.io/issues/6-build-tools/travis-ci/

booo, 2021-10-30:
As far as I can tell we build iOS apps of QGC.

DonLakeFlyer, 2021-10-30:
> As far as I can tell we build iOS apps of QGC.

That's not the case. The iOS build in the store is years out of date. It was all built by hand by Gus and pushed to the stored manually. Since Gus isn't around any more it hasn't been happening.

closed 2024-04-09 by HTRamsey