mavlink/qgroundcontrol #2947: Autoconnect: Don't keep USB links to autopilots active

LorenzMeier, 2016-03-06:
If a user pull out the USB cable it will only reconnect if he hits disconnect first. While we should have the user click disconnect for USB-to-Serial cables, we should disconnect automatically for USB links which connect directly to the autopilot, e.g. Pixhawk / Pixracer. We know that if the link is gone for those the whole thing is gone and won't return.

This will make also Firmware upgrading much less confusing.

[image]

DonLakeFlyer, 2016-03-06:
> This will make also Firmware upgrading much less confusing.

Firmware upgrade already handles this as a special case. You don't need to click Disconnect in this one case. 

Keep in mind that this indicator is "Communication Lost" not connect lost. It just means you lost heartbeats.

The fact that it doesn't auto-disconnect when you pull the cable in all cases is on purpose. I'd rather not make this change globally. For a number of reasons:
* What actually happens when you pull the usb cable differs on an OS basis. On Windows you don't actually know the difference between pulling a cable and losing heartbeats for whatever other reason.
* You can also lose heartbeats if the board crashes. If this happens then the link will just get ripped out from under you and magically disappear. Not sure if that is a good thing. Boundary case for sure.
* I think I may have just fixed one last big autoconnect bug recently. This code path is very fragile. I think it's working now and don't really want to change it anymore. The code path specifically straightens out the OS differences to make them be the same. 
* It changes the user model to be different if you are on a radio or directly connected as well as differing based on OS. I'd much rather have the user model be exactly the same no matter what you do or where you are running.

DonLakeFlyer, 2016-03-11:
Unlesss there is some really critical reason to do this I'm going to leave it alone

closed 2016-03-11 by DonLakeFlyer