mavlink/qgroundcontrol #2948: We need to wipe the map cache on upgrades

LorenzMeier, 2016-03-06:
@dogmaphobic I upgraded two machines to new QGC and got these map cache issues on both, which only resolve if you chase up the map cache dir and delete it manually. QGC should wipe the existing map cache on the next update.

[image]

DonLakeFlyer, 2016-03-06:
Isn't that a cache bug. Why clear the cache? The missing tiles should just fill in.

LorenzMeier, 2016-03-06:
They never do if you run QGC off an "old" directory. We need to wipe the cache when deploying the next stable, because all users will run into this.

LorenzMeier, 2016-03-06:
Just to be very clear: This never resolves if you don't delete the directory manually. And it never shows after. Its a transition issue from Qt's engine to our own engine.

DonLakeFlyer, 2016-03-06:
The old a new cache are completely different. The new one is SQL, I think the old one was just internet cache? I really don't think they overlap.

dogmaphobic, 2016-03-14:
I haven't tried it but I have a clue. Qt will look for its own cache before I ever get asked for a tile. That means it would find its old tile and never request it from our plugin. It doesn't explain the missing tile though.

One way or the other, keeping the old tile cache around is just wasted disk space. I will take a look and see about cleaning out an old tile cache if I see one...

dogmaphobic, 2016-03-14:
Done with #3027.

closed 2016-03-14 by dogmaphobic