mavlink/qgroundcontrol #2949: Remove FLAPS and AUX mappings from PX4 RC calibration

LorenzMeier, 2016-03-06:
@DonLakeFlyer The flaps and AUX channels would probably make most sense to have as parameter driven drop-downs in the RC calibration screen.

DonLakeFlyer, 2016-03-06:
They are on the simple mode screen. They should likely be added to Advanced as well. Instead of adding as separate element to RC Cal. This keeps RC cal clean for just setting stick assignment. Agree?

LorenzMeier, 2016-03-06:
These are linear channels, not switches and not related to flight modes. It doesn't seem to make sense to have them in the flight mode UI. They are like RPYT.

DonLakeFlyer, 2016-03-06:
So this is wrong? They should move from here to RC Cal? Leaving only Return and Kill here?
[image]

LorenzMeier, 2016-03-06:
I'm not sure I would call it wrong, but in hindsight it occurred to me that we're mixing switches with just "output channels". Would you agree that it might be more intuitive to have these in the radio tab?

DonLakeFlyer, 2016-03-06:
I kind of agree with you. But Radio Cal is already really busy ui-wise. Let me think about it.

closed 2016-03-07 by DonLakeFlyer