mavlink/qgroundcontrol #2951: Displayed and exported parameter decimal numeric precision is wrong

pkocmoud, 2016-03-06:
Params are current rounded to the third decimal. 
Per LM:

Params should just include 18 digits after the decimal point.
Firmware doesn’t round ever

 Not sure if this will also affect importing values as well.

DonLakeFlyer, 2016-03-06:
Exported precision can be 18. But displayed precision in the ui is controlled by parameter meta data. If you want more digits in the ui you need to change the meta data on the firmware side.

DonLakeFlyer, 2016-03-06:
@LorenzMeier I'm not quite sure about this one. The decimal precision in the ui is set by meta data. Whenever a value is shown to the user it will truncate to that precision.You can type in a higher precision and it will be stored internally, but when you come back to that field in the ui it will be truncated again. So I can save the values in full presicions, but the user will still never see them.

The whole reason for the precision being specified is also to work around having the floating point value blow up to many digits past decimal depending on what number you typed in due to floating point inprecision.

DonLakeFlyer, 2016-03-06:
You can't really have both: 
* A user types in a value with more precision than that meta data says and expects that to not be rounded up in the display.
* Not have the floating point values blow up to many digits past decimal in various cases.

LorenzMeier, 2016-03-06:
My understanding is that storing and loading the params is not done with max double / float precision. Is that correct or wrong? The param UI should only render these (and if in doubt incorrectly), but never change what you receive from firmware or load from a file.

DonLakeFlyer, 2016-03-06:
> My understanding is that storing and loading the params is not done with max double / float precision. Is that correct or wrong?

Correct. Just as long as you understand that the user will only ever see the rounded values in the ui.

closed 2016-03-07 by DonLakeFlyer