mavlink/qgroundcontrol #2952: Check that QGC resets the two main params for PX4 flight mode config

LorenzMeier, 2016-03-06:
If RC_MAP_FLTMODE is set to non-zero value, RC_MAP_MODE_SW should be zeroed. If RC_MAP_MODE_SW is set to a non-zero value, RC_MAP_FLTMODE should be zeroed.

If RC_MAP_FLTMODE and RC_MAP_MODE_SW are zero, the simple config should be shown (this seems to work). If RC_MAP_FLTMODE is zero and RC_MAP_MODE_SW is non-zero, the advanced config should be shown by default.

DonLakeFlyer, 2016-03-06:
That is how it works. QGC does not touch parameters unless you specifically tell it to change from simple to advanced. Which I think makes sense. Why change anything when the user didn't ask.

There is currently a bug though: If you are using old firmware without RC_MAP_FLTMODE and RC_MAP_MODE_SW == 0 it will stil go to simple mode which is wrong.

DonLakeFlyer, 2016-03-11:
This should be all fine now.

closed 2016-03-11 by DonLakeFlyer