mavlink/qgroundcontrol #2961: unable to connect to pixracer via USB

kd0aij, 2016-03-06:
1) multiple reboots of pixracer and restarts of QGC did not help
2) no hung processes starting with "q"
3) rebooting (Ubuntu 15.10) solved the problem
4) need less drastic remedy :)

DonLakeFlyer, 2016-03-07:
Without any sort of logging information there is going to be now way for me to figure out what happened here.

DonLakeFlyer, 2016-03-07:
Whever you get connection issues. Capture a log using 
```
--logging:LinkManagerLog,MultiVehicleManagerLog
```

kd0aij, 2016-03-07:
will do that next time I run into problems. I assumed this was a known issue with an existing workaround.

DonLakeFlyer, 2016-03-07:
If you have to reboot the entire OS to clear, then it more likely the port is hung up with the OS than a QGC issue.

dagar, 2016-06-08:
Have you experienced this problem again in the last 3 months @kd0aij?

kd0aij, 2016-06-08:
I don't think so.

closed 2016-06-08 by dagar