mavlink/qgroundcontrol #2969: Simulator / Flight Mode Assignments (Naming?) discrepances

GervaisK, 2016-03-07:
Users are confused, that chosen e.g. Stabilize is shown as Manual in the headline and sts. not. There´s a discrepancy between simulation and shown (assigned) FM. Other Examples:

Going to Flight Mode Page, it says MANUAL.

[image]

Changing to ACRO it shows ACRO as well in the headline.

[image]

Switching back to MANUAL, it shows STABILIZED in the headline

[image]

Switching from FLIGHT MODE 1 ACRO to FLIGHT MODE  4 ACRO it shows  MANUAL in the headline again. 

[image]

Switching from FLIGHT MODE 1 ACRO to FLIGHT MODE  5 ACRO the headline shows ACRO again.

[image]

sn0west, 2016-03-09:
I've been flying today, and there is a big problem with FM 1, 4 and 5!!! The do not take the assigned FM but take the last one that was active! This ist quite dangerous.

I did some tests and my trusted FM where 2 and 3. Unfortunaly I don't have a Log because it wont let me arme if I active the log. This is another Problem, wich I need to check before writing an Issue.

GervaisK, 2016-03-09:
This issue is not solved yet. Please don´t fly, till it is working on your desk.

DonLakeFlyer, 2016-03-09:
I checked that the parameters are being set correctly by QGC. Other than that QGC has no responsibility for changing the actual flight modes. So if this isn't workint correctly while flying it is a firmware side problem.

@LorenzMeier FYI

sn0west, 2016-03-09:
I was aware of this problem due to qgc not changing the mode in the top menue, as it is on the screenshots. So thats why I went flying and noticed this same behaviour during the flight. 
It wasn't dangerous for me because the kill switch works fine as well ;)

LorenzMeier, 2016-03-13:
@sn0west I've fixed the mapping on master now. Could you do a bench test to verify it works for you now?

LorenzMeier, 2016-03-13:
@GervaisK Closing this since fixed.

sn0west, 2016-03-15:
Oh I missed that answer, sorry @LorenzMeier. I will do a test tomorrow evening! Thanks for fixing it anyway.

sn0west, 2016-04-11:
Sorry for the late reply, but it still dosn't work!
On the following screenshots, you see the problems, FM2 is active but in the menue bar it display FM3. So What FM is now active?

I noticed some similar problem but they may have a relation to that problem

[image]
[image]
[image]
[image]

DonLakeFlyer, 2016-04-11:
The flight mode setup screen simulates flight mode seletion according the RC values, not according to the mode the Vehicle is actually in. This allows you to test you trasmitter settings while you are disarmed and not flying.

Whereas actually changing a flight mode on the Vehicle may be rejected. Hence the two can get out of sync. Using the latest QGC Daily you should get a popup when the flight mode vehicle change is rejected.

sn0west, 2016-04-11:
I'm not sure if I understand it. In my example I do change FM frome FM 1 to FM 2 using my RC. But in all four cases it shows the FM from #3 in the top bar. 
How can I control the right flight mode then? It doesn't make any sense! 
I do test with FM wich do not require any GPS lock.

DonLakeFlyer, 2016-04-11:
Can you save your parameter set to a file and attach here then I'll try to see how you have things set up. Not quite sure if the right things are happening (but very confusing) or the wrong things are happening.

sn0west, 2016-04-12:
The Taranis is set as Lorenz showed in the howto videos.
I could only upload a zip!? Shall I upload my parameters somewhere else?
[160412_snowest_PIXRACER.params.zip](https://github.com/mavlink/qgroundcontrol/files/215620/160412_snowest_PIXRACER.params.zip)

DonLakeFlyer, 2016-04-12:
Zip is good, thanks.

sn0west, 2016-04-13:
Just to be sure, I 'm the only one who noticed this problem or does someone else also have it?

DonLakeFlyer, 2016-04-13:
Ok, I can repro this by:
* Loading your parameter set onto a pixhawk board and assigning channel 5 on my radio to a dail
* I also turned off kill and return switches since I didn't have those setup
* Go to flight mode config and spin the dialog to change flight mode. You'll see that what QGC thinks the flight mode should be according to PWM value doesn't match what the Vehicle thinks it should be.
* I've checked raw rc values and they seem correct with respect to what QGC thinks the flight mode should be

Keep in mind that for FM Config QGC is doing it own calculations to show flight mode hightlight. It doesn't not use mode from Vehicle.

Not quite sure what is going wrong on the Vehicle side. I think maybe there may be some confusion over simple flight mode setup veruse advanced.

LorenzMeier, 2016-04-16:
Do you guys run master or stable when you experience this?

DonLakeFlyer, 2016-04-16:
This was on master PX4, master QGC.

sn0west, 2016-04-17:
Yes master version

LorenzMeier, 2016-04-17:
@sn0west is this an actual problem or is just the indication in QGC wrong? @DonLakeFlyer This is not using fixed PWM ranges, but rather an equal span of six slots over the calibrated range.

DonLakeFlyer, 2016-04-18:
QGC ranges are wrong then. I'll fix it.

LorenzMeier, 2016-04-18:
I fixed this - it was incomplete in PX4 and in QGC, which made it very confusing.

closed 2016-04-18 by LorenzMeier

sn0west, 2016-04-23:
The Problem is still there, maybe only a gui issue? I could go out flying to chet if the flight mode is still the right one

[image]
[image]

sn0west, 2016-04-27:
I changed the value of fm2 on my taranis and it works now. But the pwm range of fm2 seams to be realy small. I don't see the value on qgc so I can't tell from what value to what value it is. I had to press seven time on my taranis to leav the fm2 witch seams to be a realy small range. 

But it works now so everything ok.

PR #3244 (2016-04-18): Fix flight mode display for PX4. Missed reverse channel check
Fixes #2969.