mavlink/qgroundcontrol #2972: Geofences editing through the graphical interface.

MikaelFerland, 2016-03-08:
Users can edit geofences through the graphical interface without impacting the flow of a geofence check (a geofence check is done whenever a new mission is sent to the autopilot or the current position of the system is updated [https://pixhawk.org/users/geofence]).
Now, we need to find a way to handle both configurations from gui and from onboard in a way that will let each PX4 user easily configure their geofence. The onboard computer will be able to setup a geofence during flight without overiding the groundstation configuration.

DonLakeFlyer, 2016-03-09:
This needs to be supported through some sort of mavlink commands on the firmware side before QGC can implement it. Writing files to the sd card isn't a viable way to implement polygon geofence.

@LorenzMeier What's the deal with this other magic geofence stuff? Does it interfere with the parametere based geofence setup?

dagar, 2016-03-09:
It's all part of the same geofence system so they both respect geofence action (warn, loiter, rtl, terminate).

closed 2026-10-04 by github-actions[bot]