mavlink/qgroundcontrol #2974: Compass calibration not working

DonLakeFlyer, 2016-03-08:
Goes through all steps and params are set correctly but still get "PreArm: Compass not calibrated".

lucasdemarchi, 2016-03-10:
Got this today and figured out the reason.

Together with each compass device ardupilot records the "dev_id" the offsets belong to. If you start from scratch, never having calibrated the compass, the dev_id wouldn't be set.  Setting DEV_ID param doesn't work neither, because the param won't be saved.

For GCSs doing the calibration themselves they should rather call [MAV_CMD_PREFLIGHT_SET_SENSOR_OFFSETS](http://mavlink.org/messages/common#MAV_CMD_PREFLIGHT_SET_SENSOR_OFFSETS) rather than setting the params.

However... why are we doing the calibration ourselves? Could we add support for onboard calibration? I think we can maintain the steps we have in the current calibration screen, but monitor the progress ardupilot reports for compass calibration. We may need a way to set the "acceptance criteria" (the [COMPASS_CAL_FIT](http://copter.ardupilot.com/wiki/arducopter-parameters/#compass_calibration_fitness_compass_cal_fit) param) in the same screen to provide a good user experience.

lucasdemarchi, 2016-03-10:
A link with some detail about the onboard calibration: https://github.com/diydrones/ardupilot/issues/1281

DonLakeFlyer, 2016-03-10:
Yup. Working on it right now. Almost done.

The reason I did it myself is because I think the QGC mechanism is significantly better from a usability standpoint. If ArduPilot firmware supported calibration in way similar to PX4 then only visuals would be needed on the QGC side. That wasn't possible without ArduPilot firmware change. I asked a while back about support PX4 style compass cal, but didn't generate much interest.

DonLakeFlyer, 2016-03-10:
It would be awesome if ArduPilot could support a visual compass cal like PX4 does. Then QGC could get out of the compass cal business.

lucasdemarchi, 2016-03-10:
@DonLakeFlyer could you create an issue in ardupilot about that? Please give more details on why it's better than the current onboard calibration there's already in ardupilot?  I think we can improve that one allowing for better user experience but adding a third one would be a hard selling point.

DonLakeFlyer, 2016-03-10:
Now I also remember the problem with that message. It does not support a third compass. Which leaves no way for a ground station to calibration a third compass.

DonLakeFlyer, 2016-03-11:
This still isn't workinf correctly. Getting bad Z values on one comapss. need to do more testing.

closed 2016-03-12 by DonLakeFlyer