mavlink/qgroundcontrol #2975: Should support command (COMMAND_LONG) confirmation and resend

DonLakeFlyer, 2016-03-08:


dagar, 2016-06-08:
Can we do this with mavlink as is?

DonLakeFlyer, 2016-06-08:
Yes, as long as there are not multiple of the same command one the wire.

NaterGator, 2016-06-08:
Store a hash table based on command and only retry the latest?

DonLakeFlyer, 2016-06-08:
That's not quite the issue. Here it is:
1) User says to arm
2) Before ack comes back, user says to disarm
3) Ack comes back on arm/disarm command

Which one is the ack for?

NaterGator, 2016-06-09:
Sorry I have been kinda tied up in other things, does the spec not implement a sequence number included in the acknowledge? Storing the expected sequence in the hash would make it completely unambiguous if the disarm was being acknowledged or the old arm command was.

DonLakeFlyer, 2016-06-09:
Nope.

DonLakeFlyer, 2016-11-11:
Also need to replace sendMultiple with a guaranteed delivery mechanism. Right now we end up getting autopilot_version response back three times due to sendMultiple.

closed 2016-12-07 by DonLakeFlyer