mavlink/qgroundcontrol #2990: Update Android device filter to only include devices we actually support

LorenzMeier, 2016-03-10:
Users will get asked if they want to launch QGC else even though we're not supporting the devices.
https://github.com/mavlink/qgroundcontrol/blob/master/android/res/xml/device_filter.xml

DonLakeFlyer, 2016-03-11:
@dogmaphobic So leave just:
```
<?xml version="1.0" encoding="utf-8"?>
<resources>
    <!-- Everything -->
    <usb-device />
    <!-- 0x26AC / 0x11: PX4 FMU Pixhawk -->
    <!--
    <usb-device vendor-id="9900" product-id="17" />
    -->

    <!-- 0x0403 / 0x6001: FTDI FT232R UART -->
    <!--
    <usb-device vendor-id="1027" product-id="24577" />
    -->

    <!-- 0x0403 / 0x6015: FTDI FT231X -->
    <!--
    <usb-device vendor-id="1027" product-id="24597" />
    -->
</resources>
```

LorenzMeier, 2016-03-11:
No, you would need to add the 3DR Radio as well.

DonLakeFlyer, 2016-03-11:
Oh, I see it's all commented out. I'm not sure we want to filter though? We don't really know the IDs and we probably don't have string matching fallback. @dogmaphobic will have to comment on how this stuff works.

DonLakeFlyer, 2016-03-11:
Wonder what the 3DR services thing does.

dogmaphobic, 2016-03-12:
> Wonder what the 3DR services thing does.

That's where I got it from :)

dogmaphobic, 2016-03-12:
Come to think of it, it wasn't Tower. It was something else that was using a Java wrapper to c++. Tower is pure Java:

https://github.com/dronekit/dronekit-android/tree/develop/ServiceApp/src/org/droidplanner/services/android/communication/connection/usb

DonLakeFlyer, 2016-03-12:
Right but what about the manifest settings?

DonLakeFlyer, 2016-03-24:
@dogmaphobic Moving this over to you to see if you want to change anything here. I'm a bit paranoid about changing things here.

DonLakeFlyer, 2016-05-13:
@LorenzMeier, @dogmaphobic I think we should move this to 3.0. I'd rather be overly inclusive instead of getting things wrong for 3.0 by being too specific.

DonLakeFlyer, 2016-11-11:
I don't think I'm going to change this. Overly inclusive is better especially with new usb code. Less inclusive caused problems with all the different radio devices out there.

closed 2016-11-11 by DonLakeFlyer