mavlink/qgroundcontrol #2993: Font issue - QGC + Windows

marimach77, 2016-03-11:
@DonLakeFlyer Not sure if others also have the same problem. I'm using always QGC daily builds installed on Windows 7 x64. Flight Modes setup screen seems to have a font issue. Fields are too short and no PWM bands. Same issue on my other 3 computers.

[image]

DonLakeFlyer, 2016-03-11:
@LorenzMeier I can make these combo boxes a bit wider. But the meta data for these params needs to be cleaned up. No "ALL CAPS". And the "Rc Channel ##" seems a bit noisy as well. Why not just "Channel ##", or even just "##"?  It's worthwhile considering shortening as much as possible. Would "Altitude Ctl" or "Alt Control" work. Reason is that if I need to expand everything for such long wording it wreaks havoc on small screens. The other issue is that the naming in this meta data needs to match the naming and capitlization in the QGC flight modes dropdown. Otherwise users get confused. There is another issue on that somewhere.

DonLakeFlyer, 2016-03-11:
I'll adjust once the final decision on what the names of these are comes through.

DonLakeFlyer, 2016-03-11:
I'll move the meta data issue to a PX4 Issue

DonLakeFlyer, 2016-03-11:
@marimach77 Thanks for the bug report. Keep 'em coming!

closed 2016-03-11 by DonLakeFlyer