mavlink/qgroundcontrol #2994: Unit Test: Firmware upgrade

DonLakeFlyer, 2016-03-11:
Could really use a unit tests. Will be tricky to implement since it needs to simulate a serial port.

NaterGator, 2016-03-11:
[QBuffer](http://doc.qt.io/qt-5/qbuffer.html#writeData) may be a viable way to mock the serial port since both inherit from QIODevice.

DonLakeFlyer, 2016-03-11:
I think I might be able to restructure it so that I could use QBuffer for testing. Also wondering if now is the time to try moving off of QExtSerialPort which on flash uses and move to QSerialPort which everything else uses. Worried that qextserialport is so old it has not up to date bug fixes in it.

DonLakeFlyer, 2016-05-13:
This isn't doable in a way that wouldn't need a pile of mock code.

closed 2016-05-13 by DonLakeFlyer