mavlink/qgroundcontrol #2996: Reverse the order to Windows start menu item addition

DonLakeFlyer, 2016-03-11:
The issue is on Windows 10, there is a recently added item in the start menu. When you install QGC on Windows. This will show you one of the compatability mode shortcuts. Worried that users will just click on that for running QGC.

@NaterGator Do you think you could look at this? Not sure if reversing will cause other problems. Or if there is some better solution to prvent this.

NaterGator, 2016-03-11:
Sure, I'll look at how to hint the primary shortcut as the one that should be displayed after install.

DonLakeFlyer, 2016-03-11:
Thanks Nate. Hope you had a nice vacation.

NaterGator, 2016-03-14:
We actually need to set [System.AppUserModel.ExcludeFromShowInNewInstall](https://msdn.microsoft.com/en-us/library/windows/desktop/dd391568(v=vs.85).aspx). Unfortunately that looks like in requires WinAPI and we may be stuck using a grotesque script using the NSIS System plugin or compiling our own NSIS plugin DLL to manipulate the property system. Still investigating the best course of action...

DonLakeFlyer, 2016-03-14:
Nasty

DonLakeFlyer, 2016-03-14:
Something here: https://sourceforge.net/p/nsis/mailman/message/30592466/?

NaterGator, 2016-03-14:
Regrettably I'm not sure I can reproduce this issue. I have installed on Win10 and verified the property store reflects the correct values for the compatibility shortcuts, but I would be nice to get a data point from at least a Windows 7 ([reasoning](https://support.microsoft.com/en-us/kb/2745126)) and 10 user to verify this works.

closed 2016-03-17 by NaterGator