mavlink/qgroundcontrol #3002: Update to latest flight mode names

DonLakeFlyer, 2016-03-12:
PX4 
* @value 0 Manual
 * @value 1 Altitude
 * @value 2 Position
 * @value 3 Mission
 * @value 4 Hold
 * @value 10 Takeoff
 * @value 11 Land
 * @value 5 Return
 * @value 6 Acro
 * @value 7 Offboard
 * @value 8 Stabilized
 * @value 9 Rattitude

ArduPilot
Has a new QLand mode

tubeme, 2016-03-13:
If Manual = Stabilized then it is OK. So stabilized is then not a separate mode. We quit Stabilized with Acro. 

If this is their priority order it means that we cannot use Return and Hold in Acro and beyond.

Acro could be Acrobatic.

It could be like:

@value 0 Manual
@value 1 Altitude
@value 2 Position
@value 3 Mission
@value 4 Acrobatic
@value 5 Rattitude
@value 6 Offboard
@value 7 Hold
@value 8 Return
@value 9 Takeoff
@value 10 Land

And when if no GPS and Hold or Return activated can perform Land which is with highest priority.

Or without GPS lock and Hold or Return activated it could perform not pos hold but alt hold until it gets better reception. After some time without GPS or other data it could perform land.

tubeme, 2016-03-13:
In order to add more modes in between it could be a two number system for the modes like

@value 10 Manual
@value 11 Altitude
@value 12 Position
@value 13 Mission
@value 21 Acrobatic
@value 21 Rattitude
@value 31 Offboard
@value 41 Hold
@value 42 Return
@value 43 Takeoff
@value 44 Land

Even if we add some more modes in between we have to give them priority and also we have to guarantee that 41, 42, 43, 44 the emergency modes are always with higher priority.

DonLakeFlyer, 2016-03-13:
You can add a "@value 37 Foo" in the middle of the list without issue. No need to reserver space. The values don't indicate priority.

closed 2016-03-23 by DonLakeFlyer