mavlink/qgroundcontrol #3012: Add boolean support to parameter meta data

DonLakeFlyer, 2016-03-13:


DonLakeFlyer, 2016-03-13:
@dagar Let me know when you have firmware meta data in for this. Then I'll update QGC and test.

dagar, 2016-03-13:
I started using it here - https://github.com/PX4/Firmware/pull/4000

dagar, 2016-03-25:
hmm perhaps I should have waited for QGC support.

[image]

DonLakeFlyer, 2016-03-25:
@unit is only used for display to user and conversion. It does not indicate an enum or a boolean. For an enum you just add @value entries. The @unit tag should not exist for a parameter which has @value entries. To identify a new boolean type there needs to be a new @boolean tag. Boolean shouldn't be in the units. Then if @boolean, @units should not be there.

DonLakeFlyer, 2016-03-25:
Another thing to keep in mind. Enum combos only show up in the parameter editor if the meta data changes without requiring a QGC code change. Other screens like setup screens have to be coded to know that it should show a combo instead of a text field.

dagar, 2016-03-26:
https://github.com/PX4/Firmware/pull/4078 - boolean and enum usage fixed.

dagar, 2016-04-12:
Could we slip this into 3.0? The metadata change will be in PX4 stable soon and there will be many params that no longer have limits.

DonLakeFlyer, 2016-04-16:
I take a look to see how hard it will be to do.

DonLakeFlyer, 2016-05-13:
This is going to require additional firmware metadata changes in order for it to make sense to users. The QGC UI will show a checkbox control for boolean parameters. That checkbox control needs to have a standard name. I'd propose this: [X] Enabled. But that means the descriptions need to be updated to make sense with that naming. A few examples:

```
    <parameter default="0" name="INAV_DISAB_MOCAP" type="FLOAT">
      <short_desc>Disable mocap (set 0 if using fake gps)</short_desc>
      <long_desc>Disable mocap</long_desc>
      <boolean />
      <scope>modules/position_estimator_inav</scope>
    </parameter>
```

You'll have to check the "Enabled" checkbox to disable to setting. This is pretty odd.

```
    <parameter default="1" name="SDLOG_GPSTIME" type="INT32">
      <short_desc>Use timestamps only if GPS 3D fix is available</short_desc>
      <long_desc>A value of 1 constrains the log folder creation to only use the time stamp if a 3D GPS lock is present.</long_desc>
      <boolean />
      <scope>modules/sdlog2</scope>
    </parameter>
```

The long description's can cause problems as well. The wording here needs to changes to using Enable/Disable terminology as well.

dagar, 2016-05-13:
Alright I'll take another metadata pass.
My hope is that with Enabled/Disabled in QGC the short description alone should be sufficient most of the time.

DonLakeFlyer, 2016-05-13:
> My hope is that with Enabled/Disabled in QGC the short description alone should be sufficient most of the time.

Agree. You can probably clear long description for many of these.

DonLakeFlyer, 2016-05-13:
I think the way I'm going to support this is to not use a checkbox. That requires a whole new FactMetaData concept. Instead I'll turn boolean into an Enabled/Disabled enum which will be shown in a ComboBox. This reuses existing support instead of adding new stuff.

closed 2016-05-28 by DonLakeFlyer