mavlink/qgroundcontrol #3013: Update QGC param meta data, support web based download of meta data

DonLakeFlyer, 2016-03-13:


DonLakeFlyer, 2016-03-13:
The follow file would be downloaded from a location which is provided by the FirmwarePlugin:
```
{
    "fileVersion": 1,

    "parameterMetaData": [
        {
            "majorVersion": ## ,
            "minorVersion": ## ,
            "uri": uri,
        },
        ...
    ],

    "airframeMetaData": [
        {
            "majorVersion": ## ,
            "minorVersion": ## ,
            "uri": uri,
        },
        ...
    ]
}
```

* majorVersion/minorVersion - Matches values which are inside the file
* parameterMetaData.majorVerson - Matches parameter set version as specified by ArduPilot:SYSID_SW_MREV, PX4:SYS_PARAM_VER
* parameterMetaData.minorVersion - Indicates newer meta data for non-breaking change

I think the major version of airframe and parameter meta data should be tied together. Or in other words, if there is a breaking change in either airframe or param meta data it is indicated by a bump in parameter set version parameter.

DonLakeFlyer, 2016-03-13:
Would be fanstastic if we could start doing flight modes this way as well.

LorenzMeier, 2016-03-13:
Looking good. Should include a branch or git tag (so GCS can grab the right meta file based on git hash in http://mavlink.org/messages/common#AUTOPILOT_VERSION).

@dagar Might be willing to help you with uploading the airframe and parameters.xml file to AWS for each commit and updating this registry file.

DonLakeFlyer, 2016-03-13:
* If breaking change is made to parameter or airframe meta data you bump majorVersion (both in files and SYS_PARAM_VER )
* If you want newer non-breaking meta data to override older meta data at same major version level you bump minorVersion.

Semantics are not based on git hash. There is no concept of meta data set keyed to each build.

DonLakeFlyer, 2016-03-13:
Just realizing for ArduPilot this is going to need MAV_TYPE as well since parameter set version may differ between vehicle types. Which also means ArduPilot meta data needs to be broken out by vehicle type, which is not currently how it works.

Example updated

dagar, 2016-03-13:
Yes, I think he's willing.

LorenzMeier, 2016-04-27:
I have put a stopgap in Travis which pulls latest meta online before building. That will at least fix Android.

DonLakeFlyer, 2024-10-13:
Meta data and flight mode now come through mavlink with component metadata

closed 2024-10-13 by DonLakeFlyer