mavlink/qgroundcontrol #3015: (Linux) AudioOutput: espeak_Synth not triggered

shadow1runner, 2016-03-13:
Even though [emit textToSpeak(inText);](https://github.com/mavlink/qgroundcontrol/blob/dcd55081b4a5fbd5af3d6d03d77aee1394ce6009/src/GAudioOutput.cc#L107) is triggered perfectly fine (e.g. on mode change), [espeak_Synth](https://github.com/mavlink/qgroundcontrol/blob/dcd55081b4a5fbd5af3d6d03d77aee1394ce6009/src/audio/QGCAudioWorker.cpp#L154) is not triggered after the first audio output which using `sim_vehicle.sh -w` is "stabilize flight mode"). I haven't digged any further (yet), but the signal/slot connection should be there (as it's done in the ctor), but no breakpoint on `espeak_Synth` is hit after the first time.

Note that it's not only master, but also the latest release (2.9.4), here it says the audio output is "System one is now in stabilize flight mode and standing by".

DonLakeFlyer, 2016-03-13:
I changed voice output recently with respect to flight modes. I may no longer trigger a flight mode announce on first vehicle connect. Although it should.

shadow1runner, 2016-03-13:
Thanks for the fast response. I can confirm that it announces the flight mode on initial connect, but that's the first and _last_ message ever heard from QGC.

DonLakeFlyer, 2016-03-13:
@dagar You're a Linux person right? I've never been able to get speech working at all on my Ubuntu VM. Do you have time to look at this?

dagar, 2016-03-13:
I can confirm it stops for some reason. At this point it definitely works everywhere else?

DonLakeFlyer, 2016-03-13:
Just tested on OSX with a couple flight mode changes. All were announced.

shadow1runner, 2016-03-13:
And I've just tried under Windows (2.9.3 though), all worked fine.

Note that the application output after each `textToSpeak` emitting show the following line:
```
ALSA lib pcm.c:7905:(snd_pcm_recover) underrun occurred
```

NaterGator, 2016-03-17:
Sounds like espeak_Synth is never returning. Can you get a backtrace on the speech thread after it has blocked and stopped working?

shadow1runner, 2016-03-17:
Indeed, step over `espeak_Synth` never returns, here's the whised backtrace of the espeak thread:

```
Thread 9 (Thread 0x7fffc179c700 (LWP 17893)):
#0  0x00007fffefe7753d in nanosleep () at ../sysdeps/unix/syscall-template.S:81
No locals.
#1  0x00007ffff7b66b9c in ?? () from /usr/lib/x86_64-linux-gnu/libespeak.so.1
No symbol table info available.
#2  0x00007ffff7b66fab in ?? () from /usr/lib/x86_64-linux-gnu/libespeak.so.1
No symbol table info available.
#3  0x000000000050dc95 in QGCAudioWorker::say (this=0x10bacf0, inText=...) at ../../qgroundcontrol/src/audio/QGCAudioWorker.cpp:154
        text = {static null = {<No data fields>}, d = 0x3a81c70}
        espeak_size = 20
        threadInit = true
#4  0x000000000052f0ef in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QString>, bool, void (QGCAudioWorker::*)(QString)>::call (f=(void (QGCAudioWorker::*)(QGCAudioWorker * const, QString)) 0x50db9c <QGCAudioWorker::say(QString)>, o=0x10bacf0, arg=0x2f23b90) at /opt/Qt/5.5/gcc_64/include/QtCore/qobjectdefs_impl.h:501
No locals.
#5  0x000000000052f00b in QtPrivate::FunctionPointer<void (QGCAudioWorker::*)(QString)>::call<QtPrivate::List<QString>, bool> (f=(void (QGCAudioWorker::*)(QGCAudioWorker * const, QString)) 0x50db9c <QGCAudioWorker::say(QString)>, o=0x10bacf0, arg=0x2f23b90) at /opt/Qt/5.5/gcc_64/include/QtCore/qobjectdefs_impl.h:520
No locals.
#6  0x000000000052ee99 in QtPrivate::QSlotObject<void (QGCAudioWorker::*)(QString), QtPrivate::List<QString>, bool>::impl (which=1, this_=0x10baf20, r=0x10bacf0, a=0x2f23b90, ret=0x0) at /opt/Qt/5.5/gcc_64/include/QtCore/qobject_impl.h:143
No locals.
#7  0x00007ffff03463d2 in QObject::event(QEvent*) () from /opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#8  0x00007ffff23844cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /opt/Qt/5.5/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#9  0x00007ffff2389630 in QApplication::notify(QObject*, QEvent*) () from /opt/Qt/5.5/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#10 0x00007ffff0317663 in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#11 0x00007ffff03198d3 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#12 0x00007ffff036c693 in ?? () from /opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#13 0x00007ffff3c71ff7 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#14 0x00007ffff3c72250 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#15 0x00007ffff3c722fc in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#16 0x00007ffff036ca9c in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#17 0x00007ffff0315212 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#18 0x00007ffff01247bc in QThread::exec() () from /opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#19 0x00007ffff012962f in ?? () from /opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#20 0x00007fffefe6e6aa in start_thread (arg=0x7fffc179c700) at pthread_create.c:333
        __res = <optimized out>
        pd = 0x7fffc179c700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140736439371520, 3291573312334785390, 0, 140737488345407, 140736439372224, 17542112, -3291701901625726098, -3291608298351694994}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#21 0x00007fffef302e9d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
No locals.
```

NaterGator, 2016-03-17:
The libespeak1 on the build server is fairly old (1.47.11). @dagar can we get access to a more recent build on travis-ci?

dagar, 2016-03-17:
Here's a 14.04 ppa for 1.48.04
https://launchpad.net/~berkon/+archive/ubuntu/ppa/+build/6752439

shadow1runner, 2016-03-18:
Note that I'm already using the mentioned `1.48.04`:
```
dpkg -s espeak
Package: espeak
Status: install ok installed
Priority: optional
Section: sound
Installed-Size: 235
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Multi-Arch: foreign
Version: 1.48.04+dfsg-1
Depends: libc6 (>= 2.4), libespeak1 (>= 1.48.04+dfsg)
```

NaterGator, 2016-03-25:
I was able to repro this on my VM and calling espeak_Initialize with AUDIO_OUTPUT_PLAYBACK fixed this issue for me. This changed [here](https://github.com/mavlink/qgroundcontrol/commit/b7c819413c41b4fc54b1aa303dcf9527c536244a)

@DonLakeFlyer @thomasgubler any reason we can't revert back to async output?

DonLakeFlyer, 2016-03-25:
The messages step over each other if a new one comes in while an existing one is playing.

NaterGator, 2016-03-26:
Can we just call espeak_Synchronize after each call to espeak_Synth?

DonLakeFlyer, 2016-03-26:
I would assume that theoretically you can do this. But I know nothing about espeak.

NaterGator, 2016-03-30:
Should have been fixed via https://github.com/mavlink/qgroundcontrol/pull/3099, can you verify?

closed 2016-03-30 by NaterGator