mavlink/qgroundcontrol #3024: On screen emergency  button

jcooper21, 2016-03-14:


DonLakeFlyer, 2016-03-14:
This is already all implemented for Copter. Just not Plan yet.

jcooper21, 2016-03-14:
Oh right on. Once it's in there for apm plane I will test it out

DonLakeFlyer, 2016-03-14:
https://github.com/mavlink/qgroundcontrol/pull/2979

jcooper21, 2016-03-14:
For the interm is it possible to just put a pause button in QGC that makes plane loiter?

DonLakeFlyer, 2016-03-14:
Just pick Loiter from the flight mode toolbar dropdown for now.

jcooper21, 2016-03-14:
Will do. Keep up he great work.

jcooper21, 2016-03-14:
Don, I just tested the pause on today's build and it kicks the plane into loiter perfectly.

jcooper21, 2016-03-14:
With the emergency stop. Is this a setting I can disable if I want. Training people I definitely do not want them to panic and accidentally hit emergency stop instead of pause since the buttons are so close.

DonLakeFlyer, 2016-03-14:
No you cannot. You have to confirm it to make it work. So you can’t really do it by accident.

jcooper21, 2016-03-14:
Perfect

closed 2016-03-14 by jcooper21

jcooper21, 2016-03-14:
Should the emergency buttons be red in color since there emergency?