mavlink/qgroundcontrol #3029: User Issue or software limitation

jcooper21, 2016-03-14:
@Don, 

I was planning a mission with QGC and trying to insert waypoints between 2 existing waypoints and I could not figure out how to do this. I tried to renumber the point and drag the waypoints at the bottom of the screen around. Neither worked.  Also, I noticed the wind direction is unclear what angle means what since it does not match mission planners direction.

dagar, 2016-03-14:
Reordering waypoints would be extremely helpful. When creating a landing waypoint for fixed wing I've forgotten to place the approach waypoint several times.

jcooper21, 2016-03-14:
That's exactly what I was trying to do and insert a waypoint after takeoff for the plane to circle a couple times and calibrate the airspeed sensor prior to starting the mission.

dagar, 2016-03-14:
So 2 things really.
1. reorder list of waypoints
2. insert new waypoint after currently selected waypoint? I think this is usually the behaviour I want, but it might be confusing.

jcooper21, 2016-03-14:
Yes and the 3rd thing being something to reference wind directions. The angles are different than mission planner

DonLakeFlyer, 2016-03-14:
> trying to insert waypoints between 2 existing waypoints and I could not figure out how to do this.

Click the hamburger menu item on waypoint editor and select "Insert"

DonLakeFlyer, 2016-03-14:
> Also, I noticed the wind direction is unclear what angle means what since it does not match mission planners direction.

Do you mean Azimuth? Just looking at that I think I see how it's wrong. It's  not the change/difference in angle from the previous path. It's just the angle from north of the new path. It should be change in angle right?

jcooper21, 2016-03-14:
Don, I was just thinking all we really need to know is does 0 degrees mean wind is coming from the north and going to the south and 90 degrees means East to West. It's just unclear to me

DonLakeFlyer, 2016-03-14:
Where are you getting a wind value? Are you talking about Vehicle.wind.direction added to the instrument panel?

jcooper21, 2016-03-14:
Yes sir. That is what I am referring to on the instrument panel

DonLakeFlyer, 2016-03-14:
If you are using Mission Planner, where you do see this same value? I'm wondering if the spec for the data I'm using is incorrect. Need to look at MP code to see what it does.

jcooper21, 2016-03-14:
If you go to the instrument panel in mission planner and click the second mouse button on the HUD. You can select wind speed and wind direction there. After that it appears in the HUD

closed 2016-03-14 by jcooper21

DonLakeFlyer, 2016-03-14:
Can you open another Issue for the Wind problems?