mavlink/qgroundcontrol #3033: Wind speed issue

jcooper21, 2016-03-14:
I don't think the wind speed in QGC is displaying correct data. It does not seem to be actual wind direction.

jcooper21, 2016-05-28:
@DonLakeFlyer 

PX4 never displays wind direction and speed but APM does. Is this a bug?

DonLakeFlyer, 2016-05-28:
As far as I know PX4 stack doesn't support wind direction yet. Is APM showing incorrect value?

jcooper21, 2016-05-28:
I have not tested APM recently so I'm not sure if it's working on the new version. Does px4 support wind speed because that does not show up in QGC either

DonLakeFlyer, 2016-05-28:
No wind information aat all in PX4 stack,

dagar, 2016-05-28:
We'd need to implement WIND_COV  (http://mavlink.org/messages/common#WIND_COV) on the PX4 side. Internally the PX4 estimators publish a wind estimate (https://github.com/PX4/Firmware/blob/master/msg/wind_estimate.msg).

dagar, 2016-05-28:
I'll implement the PX4 side - https://github.com/PX4/Firmware/issues/4678

DonLakeFlyer, 2016-05-28:
QGC usess MAVLINK_MSG_ID_WIND

dagar, 2016-05-28:
I saw that, but common only has WIND_COV.

http://mavlink.org/messages/ardupilotmega#WIND

DonLakeFlyer, 2016-05-28:
Yeah, just realized that as well. Looks like WIND_COV is a superset of ardupilot WIND message. I can convert ardupilot WIND message to WIND_COV in the plugin and keep central QGC to WIND_COV. WIND_COV will need spec changed to allow nan in missing fields though since WIND is subset.

jcooper21, 2016-07-02:
@dagar @DonLakeFlyer  was this supposed to be in today's beta update of PX4?  I just tested the latest beta and nothing published to QGC in Android.

dagar, 2016-07-02:
You should be able to see WIND_COV in the mavlink inspector, but the rest of the QGC support hasn't been done yet.

jcooper21, 2016-07-02:
OK thanks. I was looking in the instrument panel for a reading.

booo, 2022-11-01:
As far as I can see this is implemented.

closed 2022-11-01 by booo