mavlink/qgroundcontrol #3036: connection loss when triyng calibrate the device using QGroundControl

IncognitoBA, 2016-03-16:
I have the pixhawk for qgroundcontrol, and when I try to calibrate the sensors, it shows 'connection loss'. Using Mission Planner then calibration succesful.
I found the same problem #933, but there is no solution.
Help me please!
OS Windows 8.1; Vehicle 3DR Iris+

jgoppert, 2016-03-16:
I can confirm this on windows. It is really annoying for mag calibration. When it doesn't confirm at the end the only way to set the external mag rotation is manually through the parameters.

IncognitoBA, 2016-03-17:
jgoppert, thanks for your reply. I have only one step is completed, the next shows the "CONNECTION LOST". Does it work on Ubuntu Linux?

DonLakeFlyer, 2016-03-17:
Make sure you are using latest master firmware. There was a fix made a little while back that fixed the no confirmation problem.

IncognitoBA, 2016-03-18:
@DonLakeFlyer, I tried master firmvare, but the problem remained.

DonLakeFlyer, 2016-03-19:
What problem? Communication Lost or no confirmation after calibration is complete. The change in master fixes confirmation after calibration is complete.

If you are using ArduPilot firmware the communication lost problem is not actually a real problem. It’s quirk of how ardupilot does calibration. I thought I had handled that, but I missed it for one of the calibration types. Which type of calibration are you doing? Calibration should still complete even though you get that message. Once calibration is done it should go away.

IncognitoBA, 2016-03-19:
I use PixHawk firmware. My problem is that when I turn on the RC, it shows "CONNECTION LOST" and continue the calibration impossible. Almost the same thing happens when calibrating the accelerometer and other sensors. I forgot to clarify, I'm using Windows x64.

IncognitoBA, 2016-03-21:
@DonLakeFlyer, I tried to do it on Ubuntu Linux 14.04 x64. It is the same problem.

IncognitoBA, 2016-03-23:
Solved a problem! Although the message "COMMUNICATION LOST" is necessary to continue the calibration. Alerts vehicle prompt, when the step is completed. Thank you all for the help!
P. S. and sorry for my bad English :(

closed 2016-03-23 by IncognitoBA