mavlink/qgroundcontrol #3042: Slide to arm cancel indicator confusing

LorenzMeier, 2016-03-17:


LorenzMeier, 2016-03-17:
https://cloud.githubusercontent.com/assets/749243/13810301/84a545b4-eb45-11e5-9689-f4c87e0850d6.png

In this image the cancel indicator seems to be part of the slider. I would prefer a separate circle for the cancel button, which has the proper affordable of being clicked. The current UI is extremely confusing, as you slide the arming button "onto" the abort button. That makes no sense.

LorenzMeier, 2016-03-17:
Just power off your iPhone and look how they did it. That should say it all what went wrong here.

dogmaphobic, 2016-03-17:
I dealt with this in #3037 yesterday. I too found it confusing to have the arrow pointing to a "cancel".

DonLakeFlyer, 2016-03-17:
Yeah it's kind of out of whack. Just a quick hack to get around the button problem. I still think it needs a visible cancel somewhere but I'll deal with that later

NaterGator, 2016-03-17:
Could also go the Android dialer route:
[image]

Slide left to cancel, right to confirm.

LorenzMeier, 2016-03-17:
The slide accordance of iOS is significantly better. The lack of clear indication how long / far you have to slide is terrible with Android.

LorenzMeier, 2016-03-17:
I meant affordance.

closed 2016-03-23 by DonLakeFlyer