mavlink/qgroundcontrol #3047: Lazy user preference migration when upgrading versions

NaterGator, 2016-03-18:
It would be nice if QGroundControl would attempt to retain the settings it understands when a user performs an update install.

DonLakeFlyer, 2016-03-19:
Does Windows uninstall remove the org.qgroundcontrol directory which contains the settings? If not, settings should remain untouched by uninstall or install.

jcooper21, 2016-03-19:
I can confirm I lose my instrument panel setting and comm link setting every time I update to the latest daily build.

NaterGator, 2016-03-19:
@DonLakeFlyer I guess QSettings must store user preferences somewhere different from other stuff.

We should sweep over where things are stored in each OS and determine what is and isn't fair game for the uninstall process.

closed 2026-10-04 by github-actions[bot]