mavlink/qgroundcontrol #3049: Joystick support on Android

gregd72002, 2016-03-20:
I can fly using my joystick (PS3) and QGroundControl on my Mac. However, the Android app does not show the joystick option when the controller is connected.

The controller is connected as I am able to navigate through the Android menu using it.

DonLakeFlyer, 2016-03-22:
Joystick is not currently supported on Android.

@pritamghanghas Do you ever get anywhere with this?

pritamghanghas, 2016-03-22:
actually I looked into it. QtGamePad got too many problems for our usecase.
Not worth the work. The generic event handling is already there in Qt 5.6
Android extras. In rc release it was not working when I tested it. I had
filed a bug report https://bugreports.qt.io/browse/QTBUG-51570. The bug
report has no updates yet. If this is fixed in Qt, the amount of code we
have to write is almost zero.  The bug report mentions QtGamePad. But I
mentioned QtGamePad only because it is also using that event and not
working on android right now.

gregd72002, 2016-03-25:
I can see the QtGamePad uses GenericMotionEventListener in here:
https://github.com/qtproject/qtgamepad/blob/dev/src/plugins/gamepads/android/src/qandroidgamepadbackend_p.h#L52

Are you saying the axis do not work as a result of android/view/MotionEvent?
https://github.com/qtproject/qtgamepad/blob/dev/src/plugins/gamepads/android/src/qandroidgamepadbackend.cpp#L117

pritamghanghas, 2016-03-26:
Generic mostion events never come up to Qt level. it will be something very
trivial, some merge mistake or something like that. But I haven't had time
to setup a complete android build system for Qt itself and debug and nobody
has woked on the bug otherwise.

This is the function that should work
https://github.com/qtproject/qtgamepad/blob/dev/src/plugins/gamepads/android/src/qandroidgamepadbackend.cpp#L439
for us to use the Qt event. The android code that QtGamepad has is only for
Hotplug which we anyway don't handle in QGC. For actual event handling even
the QtGamepad is using the event provided by Qt itself. Only problem is
that event is not making it through right now.

gregd72002, 2016-03-28:
I have just given it a go and created a test app that I deployed to my Nexus. Both the events (handleKeyEvent and handleGenericMotionEvent) seem to be working fine.

This is the test class I used:
http://pastebin.com/wpN902Gq

Could this be the case that the issue existed in rc release only?

pritamghanghas, 2016-03-28:
Well last time I tried qt5.6 was not released. And on RC, generic event was not working, atleast on my phone. I will give it a try again today and see if it works. If it does. I will discuss with Don and add the joystick support to QGC android. I think I have a decent Idea how to do add it with minimal change.

gregd72002, 2016-03-28:
Sounds promising. Happy to do some testing if needed to.

DonLakeFlyer, 2016-03-28:
5.6 is released but QGC is not cut over to it. It breaks a bunch of map stuff.

pritamghanghas, 2016-03-28:
so are we skipping 5.6 and jumping directly to 5.7 or when Gus gets some
time 5.6 will be fixed up. 5.7 might be a much faster release than 5.6.

DonLakeFlyer, 2016-03-28:
Not sure yet. The 5.6 problems are on me to deal with. Haven't had time to look yet. I'm tending towards skipping to 5.7 since 5.7 has Charting build in instead of on the side so we'll need to move to 5.7 as well.

gregd72002, 2016-03-28:
What difference does the qt version make to joystick support? Isn't this the same for 5.6 and 5.7?

pritamghanghas, 2016-03-28:
you can make anything work with any version. Its just amount of work that needs to be done. For joystick the way I want to do it, 5.6 will be enough if that event is working well. But there is work to be done to make QGC work with 5.6 as currently upgrading to 5.6 leads to functionality and build breakage. Its just a matter of putting in the effort to support 5.6 and probably put more effort to support 5.7 may be a month later or do it one time and jump directly to 5.7. QGC developers don't have unlimited amount of time. There are only a couple of devs like Don that do all the work. And there are ones like me who probably just try to solve their own problem and contribute about 0.000000001% of what they do.

gregd72002, 2016-03-29:
sure, understood. What's your idea of implementing the joystick support?

bartgood, 2016-04-03:
hi, I'm looking for the same. when do you expect this to be implemented?

pritamghanghas, 2016-04-03:
I will work on this on coming weekend. @gregd72002 have found out the event that I needed is working now. I think I will have some basic implementation going this weekend.

gregd72002, 2016-04-10:
@pritamghanghas do you have something we could test?

gregd72002, 2016-04-17:
How is it going with the joystick support? I had some free time today, looked into the joystick code and I think I might be able to make it work but it will take me some time. Let me know if you still working on it otherwise I will start

gregd72002, 2016-04-18:
Is there any reason why QGC code is using SDL library for the joystick and not the native Qt?

pritamghanghas, 2016-04-18:
Sorry for absense. I was away on personal work. And sorry I hven't worked
on it yet.
Reason for SDL: There was no decent cross platform code in Qt before 5.6.
If you can get rid of SDL for all platform, I think Don will merge it in
happily. But I doubt that such support cross platform support exists as of
today. Consider that the code that you are testing now should work on
Android, IOS, WIndows, Linux and Mac. SDL already cover all three desktop
platforms.

QtGamePad was supposed to be that, But they have limited mapping so much
that one would have to do a lot of work on QtGamePad itself before it
becomes usable for our purpose.

gregd72002, 2016-04-18:
Right, it looks like SDL has now joystick support on Android
https://bugzilla.libsdl.org/show_bug.cgi?id=1700

Is there any reason we are not using this? or is it just matter of testing it

pritamghanghas, 2016-04-18:
quite a big library just for joystick control and pain to cross compile and
include. I haven't looked in that direction. If it feels easy to do, You
might do some testing about code size and memory footprint. If it has
compile time macros to get rid of the stuff that we don't need, it might
make sense.

gregd72002, 2016-04-19:
thanks for the clarification. I had a look at implementing it using android jni but this is beyond my QGC knowledge. It also feels like the joystick class would benefit from re-design in terms of introducing factory or at least joystick interface.

I think I prefer to test this rather than implement. It is simply too much for my experience. But again, I might be wrong :)

DonLakeFlyer, 2016-04-19:
Last I looked the Android/SDL support was pretty unusable. You need to build a whole SDL android app based around it as opposed to using it as just a library.

gregd72002, 2016-04-21:
Right, I got some things going:
- added JoystickConfigController for all builds (it used to be excluded for __mobile__)
- joystick discovery for __android__
- joystick class for android (TODO)

https://github.com/gregd72002/qgroundcontrol/commit/c97b69b5f3dfc01ea45878022ec6ba110436ee55

https://github.com/gregd72002/qgroundcontrol/commit/877cef558a7d073627008d340b16fcd538fa1a2b

gregd72002, 2016-04-22:
Further update:
- restructured Joystick into abstract class + JoystickSDL and JoystickAndroid
- JoystickAndroid now correctly receives all the gamepad events

https://github.com/gregd72002/qgroundcontrol/commit/7af1f46c164370da6ae236d61ce2925776ad2b73

However, I'm having problems testing it properly as the UI does not show the Joystick menu in the properties despite having discovered a Joystick. What am I missing?

gregd72002, 2016-04-22:
I think I am done in here as far as I could. The code is there but it looks like the calibration fails on Android with segfault for no obvious reason to me. All works fine on OSX.

Can someone help me with this?

gregd72002, 2016-04-23:
I have given it another go and found the problem. It should be now all working.

This builds on Android and OSX (tested). And joystick works on both.
https://github.com/gregd72002/qgroundcontrol/tree/AndroidJoystick

The limitations:
- only one joystick is supported on Android. This can be changed but would need some further re-design if needed. But for now all should be working fine.
- the UI for the Joystick looks a little bit messy but useable

gregd72002, 2016-04-26:
Pull request created: https://github.com/mavlink/qgroundcontrol/pull/3286

nanthony21, 2016-08-14:
Did this ever get fixed? I can't seem to get android QGC to recognize that a controller is plugged in.

gregd72002, 2016-08-14:
@nanthony21 it is fixed in a pull request. This has not been merged yet. You will need to compile it yourself with this pull request https://github.com/mavlink/qgroundcontrol/pull/3596

DonLakeFlyer, 2016-08-15:
Keep in mind that pull request requires a newer version of Qt which is going to break displaying of items on the map.

nanthony21, 2016-08-15:
Hmm, ok. I might get around to it. Is there any estimate on when this will get merged into master?

DonLakeFlyer, 2016-08-15:
Need to wait until Qt 5.7.1 is released. Haven't seen a date on that yet.

pritamghanghas, 2016-08-16:
Don, If the code is just enabled forAndroid. 'The map items not showing up'
doesn't happen with mobile devices even with 5.7.0. I haven't looked at
full extent of code. But feel we might be alright.

Regards,
Pritam

DonLakeFlyer, 2016-08-18:
Duh, good point. So anyone that wants to build with the pull could give it a try. Still want to wait to we move all OS to 5.7.1 to merge though otherwise if requires too much separate testing.

NaterGator, 2016-08-18:
I also cherry picked the QML map fix and built Qt 5.7 with PDBs for release mode to so I can do proper release debugging and profiling. 

Since the CI system already downloads tarballs for Qt there is nothing stopping us from rolling our own Qt 5.7 builds with the map fix. I haven't heard much about the 5.7.1 release schedule.

pritamghanghas, 2016-08-19:
I had also cherry picked and tested with 5.7 custom build and it had worked
fine. For people who build from source it will cause too much support
requests. I think bumping to 5.7 and asking people to build their own Qt to
have right functionality will be nightmare to support. Lot of people who
are not that involved with Qt and Qt Creator build QGC from source and they
will ask for a lot of hand holding.

Regards,
Pritam

rvdgracht, 2017-06-16:
Hi Guys!

I wrote custom firmware for my custom hardware and I'm trying to use this with QGroundstation. I've got it working with my modified Xbox controller on my linux system. But I cant get joystick working on my Android device. I've tried stable 3.1.3 and todays daily build without success. According to this thread, it should be operational for a while. Or is it stil work in progress?

gregd72002, 2017-06-16:
do you know if the system recognises it at all? i.e. are you able to navigate your android menu using the joystick?

DonLakeFlyer, 2017-06-16:
@gregd72002 I'm going to close this since you implemented the support already. A new Issue can be entered if the discusssion leads to a real problem.

closed 2017-06-16 by DonLakeFlyer

Mocart96, 2017-11-27:
Hello. Can you let me know what is status joystick implementation for android devices, please ( is this dopped or on hold ? ). I would like navigate my copter using joystick instead of virtual sticks only.
I did short test a few days ago and my joystick works well with windows 10 ( FlySky FS-I6S in joystick mode - sample movie: https://youtu.be/bfCuC1I42DQ) but it does not work with my android tablet (Samsung Tab4). 
FS-I6S is recognized by android without any problems. I can easy navigate android menu or move between desktops. I did record also same test for android, it is available there: https://youtu.be/LvVuqIIMVmM
I hope both movies shows very well how it works now and question is : can it be finally supported.
One more, in the case of mobile app the is a lack of joystick tab under settings menu.

pritamghanghas, 2017-11-27:
Doesn't look like that you looked for joystick in the menu after connecting
it. May be it is there and needs configuration.

Regards,
Pritam

pritamghanghas, 2017-11-27:
> In there case of "desktop"/ Windows 10 application after connection to
> qground control there is a separate tab called "joystick" where you can
> select an input device.
>
> For Android there is lack of this tab, and you can change input only from
> parameters tab. Please check my print screens.
>
> Are you able to share with me screen with joystick tab active ? In the
> case of desktop menu is not dynamic. If there is no joystick connected tab
> is grey and stay active once joystick is connected.
>
> For Android joystick tab does not exist. Can someone help with it ?
>
> [image: 1_tab]
> <https://user-images.githubusercontent.com/3898523/33291167-38d40638-d3c5-11e7-804a-b49f8b91394f.jpg>
> [image: 2_parms]
> <https://user-images.githubusercontent.com/3898523/33291176-3edf6cde-d3c5-11e7-9f9a-2025c94edda9.jpg>

can you try daily. There was a problem in last stable release, which will
not allow joystick on android to work. Test daily once to confirm.

Mocart96, 2017-11-28:
[image]
Test is done using daily build. Find attached screen with build version. QGC on Android Kitkat.

Full test record available on YouTube : https://youtu.be/8LJ_nvPz1tQ
Scenario: To arm and start motors using usb connected joystick. No RC connection
Parms are:
Com_RC_in_mode=1 ( Joystick/No RC check ) 
Safety: Lost on RC & Lost on data - both are DISABLED
Joystick - supported by OS
Virtulal joystick checkbox - unchecked
Result : not success.  There is lack of joystick tab under settings menu. There is no possibility to arm and run motors using USB connected device.

I did make one another test with my Samsung Galaxy S7 ( with Android 7 ) and results are the same.

What can I do else to make it running ?

pritamghanghas, 2017-11-28:
You will have to gather console logs after enabling joystick logging,
otherwise there is no way of knowing what is going on.

Mocart96, 2017-11-28:
What does it means ? I understand I can save logs from mobile when qground control runs because there is no "joystick" tab under mobile menu like it exist for desktop ( screen attached ). 
I cannot select/deselect joystick, From mobile level I can setup only "Com_RC_in_mode" parm.
Is this fine to you to have logs saved during QGC launching process ? 

[image]

pritamghanghas, 2017-11-28:
Yes, logs saved during launch are fine. we only need to see whether it
recognized the joystick or not. you will have to take logs from android, If
you are using prebuilt binaries. There is console tab in application
settings menu. I think you can save the logs from there. Even if there is
no joystick shown, the logs will contain what went on. Don't forget to
enable joystick logging from set logging button.

rvdgracht, 2017-11-29:
I'm running qgc on a tinkerboard (Linux) and encountered an issue when using my xbox wireless controller receiver which might be related. The receiver supports a maximum of 4 controllers, and registers them (all 4) at once when a single (first) one is paired. 

In /src/Joystick/JoystickManager.cc; _updateAvailableJoysticks() gets called 4 times for a SDL_JOYDEVICEADDED event. All 4 joystick devices have exactly the same name. This results in reassigning the _activeJoystick varible 4 times, which makes qgc look for input events from the 4th joystick which not connected (in my case).

Mocart96, 2017-11-29:
I was trying to capure logs today.  First of all following groups are selected:
[image]
Base on following screens It`s look like some valuse are logged. You can see there Fly sky controller is available in those logs
[image]
and now I have a problem how and where I can find those logs on android device. Base on following page there is no details where they are kept https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html
I found Qground Control folder on my android device but there are not logs. Screen below ( sorry for local language , will change it next time ) 
[image]
Additionaly I have installed 2 version of qground control now. Regular and daily build.
Could you advice where can I find daily builds logs on my device ???

Mocart96, 2017-11-29:
I dont know how to save those logs from android devices. In meantime let me share some printscreens from those logs. You can see there that FS-I6S is recognized by qground control as joystick but joystick setup screen is not available. I tested also my sony ps-4 pad and results are the same. Android version recognize it also but do not display joystick setup screen.
I`m waiting for any help how I can download from android any log file ( where is correct folder etc. )

[logipix.zip](https://github.com/mavlink/qgroundcontrol/files/1515694/logipix.zip)

Mocart96, 2017-12-01:
Hello, Do we stop to discuss about it? I little confused with this joystick support. As I understand Ps4 pad was approved and tested device some time ago. My Fly Sky is show in the logs similar to PS4. Can someone share sample screen of joystick page when ps4 pad is connected to QGC mobile, please ? Maybe someone can help me to grab those logs form my android device also if previous screens are not helpful ?.
Regards,
Adam

pritamghanghas, 2017-12-01:
will you be able to check a custom build. I am away from a dev setup. But I
have a build that I know works for me.

Mocart96, 2017-12-01:
ok, cool. I can test any version if you can share it with me. 
As I said I have 2 controllers now where one is officialy supported by desktop app.

pritamghanghas, 2017-12-01:
try this http://senserobotics.com/qgc26.apk. This is mostly my hobby build,
might have lot of other cruft.

Mocart96, 2017-12-01:
Find attached screens from my quick test. I did not check everything step by step but with this version Fly Sky controller is available in QGC on boot. 
Test was done using Samsung Galaxy 7 and now I can say I never saw this screen in regular builds. 
I will be testing it today but now we know there is something wrong with regular and daily builds. 
This screen does not exist in other versions of QGC. Once I test Everything I will share results but please take a look on regular build why it`s missing

[image]
[image]

Mocart96, 2017-12-01:
There is outcome of my tests:
1. scenario 1: Com_RC_in_mode=1 (Joystick/No RC check). https://youtu.be/aKBqeVmfC9o
it works amazing ;)
2. scenario 2: Com_RC_in_mode=2. (Virtual RC by Joystick). Screens are attached. I cannot arm morors due note: " manual throttle non-zero). This is very starnge becuse I didnt make any changes to joystick setup camparing scenario 1. Do you know where can I find root of this problem?

pritamghanghas, 2017-12-03:
I think you are talking about px4 firmware. I have no experience with that.
The fix for joystick is in master. You can use daily builds. I just don't
want you to get stuck with something that is different in my build. Because
the build you have is very specific and not well tested.

Mocart96, 2017-12-10:
Hi. What does it means - its in master? If you check some posts above I did test daily builds ( screens are attached ) and well working joystick support is available in your custom version only(http://senserobotics.com/qgc26.apk).

pritamghanghas, 2017-12-11:
> qgc26.apk).

It shd be fixed in daily now. That is what I meant when I said it is in
master now. From that day onwards daily is fixed.

Mocart96, 2017-12-12:
Yes ! Works amazing with daily build 👍 
I have 2 additional questions:
1. When it will be available in regular version ?
2. Does it work with ardupilot firmware or px4 only? ( you are right I have px4 installed currently )

[16 more replies]