mavlink/qgroundcontrol #3050: TCP connection broken

dagar, 2016-03-21:
It will connect, but no param sync.

DonLakeFlyer, 2016-03-22:
I think this has been broken for ages.

dagar, 2016-03-22:
Do you know if it ever worked?

dogmaphobic, 2016-03-22:
When the first ESP8266 modules came up, some time last year, I tried it and it was working (those modules only handled TCP). Not well but it worked. I don't have anything (normal) to test with though...

DonLakeFlyer, 2016-03-22:
Last time I tried it a while ago I seem to remember it crashing. I think the signalling was out of whack.

NaterGator, 2016-03-22:
Would testing with mavproxy be suitable? If so @dagar and/or I can probably knock this out

DonLakeFlyer, 2016-03-22:
Last I tried was TCP connection to ardupilot SITL. That didn’t work.

DonLakeFlyer, 2016-03-23:
> Would testing with mavproxy be suitable?

Yes I think it would. Can you take a look?

NaterGator, 2016-03-25:
The primary issue I have found so far in playing around with this is that link code is not thread safe. Links like TCPLink and UDPLink spawn their own event loop thread and move themselves to that thread, then functions like Vehicle::_sendMessageOnLink invoke LinkInterface::writeBytes directly. Are you comfortable with me making this interface thread-safe before the 3.0 release?

dogmaphobic, 2016-03-25:
I never took the time to clean up the link interfaces. Threading is an issue with all of them. It's not just TCP. We're having quite a bit of lost packets with UDP as well and I can only assume it's related to the weird threading model used by those classes.

DonLakeFlyer, 2016-03-25:
Fixing the threading is fine. But since Vehicle's all run on the main thread you won't actually hit a threading problem with message sending. So unlikely to be causing the TCPLink problems. The case where it can happen is when mavlinkprotocol sends a message to the vehicle internally to it without going through Vehicle. This happens with PING and multiplexing. So a problem, but also not likely the issue with TCPLink people are seeing.

NaterGator, 2016-03-25:
That's actually incorrect. For example with TCP sockets sending on the main thread invokes [QAbstractSocket::writeData](http://code.qt.io/cgit/qt/qtbase.git/tree/src/network/socket/qabstractsocket.cpp?h=v5.5.1#n2515), which in turn calls [QSocketNotifier::setEnabled](http://code.qt.io/cgit/qt/qtbase.git/tree/src/corelib/kernel/qsocketnotifier.cpp?h=v5.5.1#n233) which cannot be invoked from a thread other than the thread that owns the socket. Since moveToThread in the ctor changes ownership the object that subsequently creates the socket, the socket is owned by (and run in the event loop of) a thread other than the main thread.

Let me take a pass at fixing the LinkInterface implementation and we'll see how many problems that alleviates.

DonLakeFlyer, 2016-03-25:
Right, that's the part that is broken. Now I remember I had seen this before. For threading I was talking about issues with multiple callers to writeBytes coming in at the same time. That doesn't really happen in the current code.

NaterGator, 2016-04-07:
This should have been fixed by #3079. If not please reopen.

closed 2016-04-07 by NaterGator