mavlink/qgroundcontrol #3055: Check direct connect of PX4Flow to QGC

DonLakeFlyer, 2016-03-22:
Reports of issues

proficnc, 2016-03-24:
[image]
[image]
[image]
[image]

When using firmware version.... which is reccomended for ardupilot.
http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/px4flow-klt-06Dec2014.zip

DonLakeFlyer, 2016-03-24:
That firmware is reporting itself as MAV_AUTOPILOT_PX4 which is the id for PX4 Firmware. This causes QGC to think it's a Pixhawk running PX4 Firmware. Which then in turn causes all sorts of strangeness. Since there isn't a specific entry for FLOW firmware in the MAV_AUTOPILOT enum it should be reporting itself as MAV_AUTOPILOT_GENERIC not MAV_AUTOPILOT_PX4. The standard flow firmware which QGC installs comes from here: http://px4-travis.s3.amazonaws.com/Flow/master/px4flow.px4. It works correctly with respect to reporting MAV_AUTOPILOT enum values.

LorenzMeier, 2016-03-24:
The flow binary you're referring to is stale and should not be used.

closed 2016-03-24 by LorenzMeier