mavlink/qgroundcontrol #3056: CPU load 100% on disconnect

julianoes, 2016-03-22:
The CPU load seems to go to 100% for one core when a serial connection gets disconnected. It is independent of auto-connect settings. A restart resolves the problem, and gets it back down to 2.4%.

This is on 2.9.4 on Ubuntu.

DonLakeFlyer, 2016-03-22:
Disconnected by pulling USB?

julianoes, 2016-03-22:
Correct, using a Pixhawk.

DonLakeFlyer, 2016-03-23:
For me, when you click disconnect the CPU goes back to normal. Still not sure what is going on.

LorenzMeier, 2016-03-23:
People don't click disconnect. They pull the cable and then get angry because 15 mins later their laptop battery has drained like crazy. The disconnect button has a terrible user model. A user never connects anything, so why would he think of disconnecting something he never connected?

DonLakeFlyer, 2016-03-23:
I explained the reason for disconnect to you already in another issue.

LorenzMeier, 2016-03-23:
I understand and appreciate it. However, the various user issues resulting from it won't go away if I understand the design choice. They don't understand it and just see that the app pulls 100% CPU load.

One option would be to make the disconnect button graphically much more visible (yellow / orange) and to bring the application to the front (equivalent of "make it jump in the dock") if a disconnect happens.

If you expect users to click disconnect, then let's help them to get there. Right now its extremely easy to miss.

DonLakeFlyer, 2016-03-23:
You seem to think I'm not going to fix the CPU problem? You don't want to make Disconnect too annoying. It can come and go when you are running over WiFi with a weak signal.

LorenzMeier, 2016-03-23:
I guess that's the miscommunication, yes.

julianoes, 2016-03-23:
@DonLakeFlyer Can you just make the reconnect tries slower so it pulls less CPU? Or is there more to it?

DonLakeFlyer, 2016-03-23:
There is no reconnect trying going on. No such thing with autoconnect. Port is still open.

julianoes, 2016-03-23:
I see, do you know what it is that is cycling?

DonLakeFlyer, 2016-03-23:
Nope, not yet. Wondering if its in the serial port code somewhere.

DonLakeFlyer, 2016-03-24:
So far all things pointing to Qt SerialPort code which is bad since we don't ahve control over that.

LorenzMeier, 2016-03-24:
We should fix the issue on the conceptional side. Since we auto-connect to new links, we can afford to auto-disconnect links after a short timeout (5-20 seconds). If the link comes back we can re-connect. If it doesn't the CPU load only increased for a short amount of time.

And most importantly, users who pull out the USB cable and don't click disconnect can actually re-connect to the autopilot. As it stands right now it looks "broken" for them.

I still think the disconnect button should be yellow. The current subtle approach really doesn't work well with users.

NaterGator, 2016-03-24:
FWIW: I do not see this CPU load on windows. Most likely this is Qt blocking on a QWaitCondition/QSocketEventNotifier or similar synchronization primitive. A stack trace should point to exactly where things are deadlocked and help untangle this. 

Further I see no reason why we can't discern the difference between a non-responsive (but valid) link and a physical removal of the link. I am certain it is possible to detect on Windows and would be shocked if it is not possible on Linux or OS X.

DonLakeFlyer, 2016-03-24:
> Further I see no reason why we can't discern the difference between a non-responsive (but valid) link and a physical removal of the link.

This can't be done with UDP connections. There is no way to know if the vehicle has been turned off or it is out of range and/or lost communication for whatever reason. If you're running a WiFi connection  with a  weak signal it's pretty common for you to lose communinication and have it come back. This happens on my PixRacer all the time. I take off with good WiFi signal. Then I fly out of range for random amount of times and then come back to land and get the signal back. It's going to be really annoying if that disconnects automatically and starts asking me to save telemetry logs and what not right in the middle of a flight.

On Windows when you pull the plug on a serial port, the port does not signal that it has gone away. On Linux/OSX it does. This is all happens in the QSerialPort code. It might be doable to detect link unplug on Windows with some custom windows code.

It's the way it is now to keep the semantics the same for all link types across all OS. It's not this wayh on accident, it's this way on purpose.

Like I said in the other thread a few weeks back. Auto-connect is finally working fairly reliably. I don't want to change the semantics of it right now (if even possible). It has been very fragile up to this point. I'd rather wait until we get 3.0 out the door to change anything around it.

DonLakeFlyer, 2016-03-25:
This still looks to be hanging up in the Qt serial port code. So as far as I can tell I can't fix from the QGC side. I still haven't given up yet on finding the fix, but my hopes are not high. But also trying to make Disconnect more prominent so people know to click it in the mean time. Auto-Disconnect will come in 3.1.

NaterGator, 2016-03-25:
Let me see if I can work with @dagar to repro this in Ubuntu (regrettably all I have are Ubuntu VMs) to see if I can offer any insight.

DonLakeFlyer, 2016-03-25:
I tried pulling stack traces from OSX but they weren't very helpful.

dagar, 2016-03-26:
I'll try to reproduce.

dagar, 2016-03-27:
oprofile of pixhawk connected over usb then pulled
Ubuntu 14.04 with QGC master

    CPU: Intel Sandy Bridge microarchitecture, speed 3.2e+06 MHz (estimated)
    Counted CPU_CLK_UNHALTED events (Clock cycles when not halted) with a unit mask of 0x00 (No unit mask) count 100000
    samples  %        image name               symbol name
    267917   21.9249  libglib-2.0.so.0.4002.0  /lib/x86_64-linux-gnu/libglib-2.0.so.0.4002.0
    199682   16.3409  libQt5Core.so.5.5.1      /home/dagar/Qt/5.5/gcc_64/lib/libQt5Core.so.5.5.1
    172940   14.1525  no-vmlinux               /no-vmlinux
    65347     5.3476  libpthread-2.19.so       pthread_mutex_lock
    57369     4.6948  libpthread-2.19.so       pthread_mutex_unlock
    31974     2.6166  libdbus-1.so.3.7.6       /lib/x86_64-linux-gnu/libdbus-1.so.3.7.6
    27024     2.2115  libX11.so.6.3.0          /usr/lib/x86_64-linux-gnu/libX11.so.6.3.0
    20910     1.7112  libxcb.so.1.1.0          /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
    20222     1.6549  libc-2.19.so             __dcigettext
    20000     1.6367  libc-2.19.so             _int_free
    19990     1.6359  libc-2.19.so             malloc
    19258     1.5760  libc-2.19.so             _int_malloc
    18842     1.5419  libc-2.19.so             malloc_consolidate
    18112     1.4822  i965_dri.so              /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
    17446     1.4277  libc-2.19.so             __GI_strcmp
    16607     1.3590  libQt5Widgets.so.5.5.1   /home/dagar/Qt/5.5/gcc_64/lib/libQt5Widgets.so.5.5.1
    14420     1.1801  libpthread-2.19.so       pthread_rwlock_rdlock
    14304     1.1706  libpthread-2.19.so       pthread_rwlock_unlock
    12254     1.0028  libQt5Gui.so.5.5.1       /home/dagar/Qt/5.5/gcc_64/lib/libQt5Gui.so.5.5.1

dagar, 2016-03-27:
backtrace

    #0  0x00007f851897061a in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
    #1  0x00007f8518970979 in g_mutex_lock () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
    #2  0x00007f851892e0b0 in g_main_context_acquire () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
    #3  0x00007f851892eea5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
    #4  0x00007f851892f0ec in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
    #5  0x00007f851502aa87 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/dagar/Qt/5.5/gcc_64/lib/libQt5Core.so.5
    #6  0x00007f8514fd3212 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/dagar/Qt/5.5/gcc_64/lib/libQt5Core.so.5
    #7  0x00007f8514fdad3d in QCoreApplication::exec() () from /home/dagar/Qt/5.5/gcc_64/lib/libQt5Core.so.5
    #8  0x000000000044a05e in main ()

dagar, 2016-03-27:
All the activity is outside of qgroundcontrol so I'm not quite sure what to look for. Any pointers?

DonLakeFlyer, 2016-03-27:
Same problem when I tried to debug this on OSX. Couldn't make any sense of what was going on inside the OS. All greek to me!

DonLakeFlyer, 2016-04-12:
We've got a while before we get 3.0 out the door. So I';m going to take a stab at auto-disconnect now, since I think there is enough time to stabilize it out. Rather get important user model like this right before such a critical release.

dagar, 2016-04-13:
Resolved.

julianoes, 2016-04-14:
Thanks guys, looking forward to using it!

closed 2016-04-19 by DonLakeFlyer