mavlink/qgroundcontrol #3057: Non-display of component Parameters with multiple components

tonymtsai, 2016-03-22:
With multiple components reporting parameters on the startup of the program, the components that take a bit longer to respond ~1sec do not appear on the list.  This happens when the first few components to respond have very few parameters ~1-2.   If I pad out the first few components with more components ~8-9, the slower-to-respond component then appears on the parameter list.

Looking at the ParameterLoaderLog, the program always sees the slow-to-respond component with "Seeing component for first time, id: 1 parameter count: 390" and all subsequent parameters.  But they are not displayed in the program window unless the first few components are padded. 

Logs attached. cid#1 is always slower to respond. cid#240, 241, 26 has fewer parameters.
[parameter_not_padded.txt](https://github.com/mavlink/qgroundcontrol/files/184843/parameter_not_padded.txt)
[parameter_padded.txt](https://github.com/mavlink/qgroundcontrol/files/184844/parameter_padded.txt)

DonLakeFlyer, 2016-03-23:
Can you wait to go to Parameter Editor until all components are done downloading params. Check by using log output. Or just wait fora  minute or so till you can be sure all parameter load is done. And then once everyting is done, go to parameter editor and see if all components show up? I think I know what is going on. Just want to check.

tonymtsai, 2016-03-23:
Yeap so I've waited to go into the parameter editor until all the params are loaded but the behaviour is the same.  The slower to respond components do not show up.  Is there another log type you want me to post?

DonLakeFlyer, 2016-03-23:
Can you save a telemetry log of your vehicle connect and attach to here?

tonymtsai, 2016-03-23:
I can't see any log files on the pixhawk.  I've run qground control with --logging:full,LinkManagerVerboseLog,ParameterLoaderLog.  

So the same the first few components appear on the list.  The pixhawk takes about 5 seconds to pump out all its params and doesn't show in the parameter editor.

[full_log.txt](https://github.com/mavlink/qgroundcontrol/files/186836/full_log.txt)

DonLakeFlyer, 2016-03-23:
Telemetry logs are from QGC not pixhawk. When you disconnect from the vehicle you should be prompted to save the telemetry log by QGC.

tonymtsai, 2016-03-23:
Right,  :) Just had to enable saving even if it was not armed.
[log.mavlink.zip](https://github.com/mavlink/qgroundcontrol/files/186870/log.mavlink.zip)

DonLakeFlyer, 2016-03-24:
@LorenzMeier This problem is caused by a hole in the parameter protocol. I've known the hole was there for a while now but hadn't run into problems because of it yet. The issue is that QGC needs to know when the initial parameter load is complete. The parameter request protocol doesn't really handle multiple components in way that tells a ground station how many components there are. If a component finishes sending all of its parameters before another component sends its first parameter QGC signals load complete. Since it had no idea that more components are coming. I'll see if there is some way to handle this from the QGC side but it's going to be tricky.

LorenzMeier, 2016-03-24:
The fix is simple: Do not consider the param load complete for 2 seconds after the parameter request has been sent (just take the timestamp on sending and do not signal complete if less than 3 seconds have elapsed).

You also *have to* create a single-shot ```QTimer``` which signals complete if the list is complete after 3 seconds (because if no transfer happens after 2 seconds, it will never signal completeness else).

DonLakeFlyer, 2016-03-24:
That's going to delay QGC vehicle connect by another 2 seconds. Maybe that's ok. I was trying to prevent slowing things down even further. I may adjust the concept so that it understands whether it is missing the default component parameters. If it does have default component then signal initial pass complete. Then update for additional component that may come in later. I need to move some things around to be dynamic as new components come it to make it all work.

DonLakeFlyer, 2016-03-24:
@tonymtsai Do you have the ability to build QGC yourself? Would like you to test a change I'm putting together.

tonymtsai, 2016-03-25:
Yeap I do. Will test it out and let you know. Does this fix extend the wait time for any components or just the main component?

DonLakeFlyer, 2016-03-25:
It's kind of hard to describe how it all works internally. There are multiple cases of timeouts in play. It will not add any extra dead wait time to the QGC parameter load complete detection. QGC will signal parameter load complete (thus showing you setup ui) once it gets the params for the main/default component. If components are still coming in after that they will be added automatically. It is possible to get into parameter editor before all non-default components are loaded. In that case, once those additional components finish loading params, If you switch out/in to parameter editor they will show up. Like I said it's kind of complicated. It's pretty much as good as it can get given the protocol.

tonymtsai, 2016-03-29:
Had a chance to test the new code and it works well.

DonLakeFlyer, 2016-03-29:
Thanks for letting me know.

closed 2016-03-29 by DonLakeFlyer

tonymtsai, 2016-03-29:
Now that that works, I just have a question about the sending of the _HASH_CHECK value.  After the pixhawk (with PX4) sends this value to the groundstation at the start, the groundstation sets this value back on the pixhawk.  This causes the pixhawk to stop sending the rest of the parameters back to the groundstation. The groundstation will however still load all the parameters without issue.  

I just want to check that this is the standard behaviour intended.

DonLakeFlyer, 2016-03-29:
This is standard behavior. Ths GCS sends _HASH_CHECK back to the Vehicle to signal to it that it can stop sending params since it is going to use it's cached values.