mavlink/qgroundcontrol #3058: Map & Mission Issues with current master

tops4u, 2016-03-22:
Map Handling in Mission, Flight Data and Cache Handling does not work well on Mac (Mac Book with Touchpad). Pinching in order to Zoom in or out is erratic. One Pinch action usually works when starting to Zoom more in or out the Zoom factor is often reset to the initial value before starting to zoom in the desired way again. Moving Maps is also not working well... sometimes it works when using 2 fingers or by click'n move or by no means at all. 

Using the current Daily Master Build Icons do not show up in the Mission Planing leaving the user without any clue about Waypoints or other actions.

DonLakeFlyer, 2016-03-23:
There are bugs in Qt with respect to touchpad handling. We've tried to work around them, but there doesn't seem to be a way. This will be fixed once we move to a new internal architecture on desktop versions. For now, you need to make sure to not use two finger operations from the touchpad. If you do it will get stuck in a mode which doesn't allow the rest of the functionality to work.

I'll keep thinking about workarounds for this. The architectural change is likelt a few months away.

> Using the current Daily Master Build Icons do not show up in the Mission Planing

Do you mean that your vehicle has a mission on it and when you go to Plan the mission doens't show up? You need to load it from the vehicle manually. Need to reconsider whether this should happen automatically.

NaterGator, 2016-03-23:
@DonLakeFlyer can you bring me a bit more up to speed on where the issues lie and what fixes were attempted? I may be able to help iron out some of the Qt quirks.

DonLakeFlyer, 2016-03-23:
When you start a two finger gesture (pan, zoom) on the mac touchpad (this is a Mac only problem as far as I can tell) it gets stuck and never ends the gesture. From then one your map is hosed. You can single finger touch around and you may eventuaIly gets things back to normal. I tried to figure out how to turn off the gestures and implement pan/zoom myself but couldn't get the gestures to go away completely. It would still get stuck. This problem only occurs if you are using the Map in a QQuickWidget. If you run a fully native Qml desktop app the problem goes away.

tops4u, 2016-03-23:
@DonLakeFlyer when planing a new Mission no Waypoint Markings at all showup on the map. This is on actual Head on Master.

DonLakeFlyer, 2016-03-23:
Can you attach a screen shot? Does anything show up when you add waypoints. Right hand editors? Keep in mind that if you are stuck in the mode where zoom is stuck, clicking in the map to add waypoints won't work.

tops4u, 2016-03-23:
Yes of course. Here it is. I just started up QGC, placed a few Points on the default Location and even pressed "Center Map". As you can see there are several points and they also show up in the Status Bar on the bottom, but not on the Map itself. 

[image]

DonLakeFlyer, 2016-03-23:
What do you have you offline map editing set to? ArduPilot? If you don't know what that is, then that is what it will be.

If you zoom out, the waypoints aren't in some random location?

tops4u, 2016-03-24:
@DonLakeFlyer  AFAIR I did set thie to PX4 as I wanted to test various things, I also activated the experimental Support for survey (for which I never managed to draw a Polygon - but that's a different thing). I did not zoom out but I clicked in the actual visible Map to set the points, also as mentioned clicking Move Home to Center of Map did not reveal the Icon. Tested on a MacBook and on an iMac - same behaviour (as well as same User 8-) )
Stable 2.9.4 did not have this problem.

DonLakeFlyer, 2016-03-24:
The problem is that the build system which drops daily builds for OSX uses brew install to get the version of Qt onto travis containers. Since 5.6 was just released "brew install" is installing 5.6 Qt which QGC doesn't work with yet. It needs to be using Qt 5.5. I'm going to need to create a custom OSX Qt 5.5 package to use for OSX just like I did for iOS builds.

@dagar Just fyi to you. I'll fix this up.

tops4u, 2016-03-24:
I'd more than happy if it is fixed in the next stable release.

DonLakeFlyer, 2016-03-24:
Should have it fixed up today.

DonLakeFlyer, 2016-03-27:
FYI: Bug where mission items where not showing up is fixed now, finally!

DonLakeFlyer, 2016-05-25:
Workaround for Pinch zoom craziness is in place.

closed 2016-05-25 by DonLakeFlyer