mavlink/qgroundcontrol #3067: Tuning sliders (e.g Position control sensitivity) don´t keep their position, return to left att.

GervaisK, 2016-03-23:
No problem to adjust them, but once one leaves the tuning page and return again, they are back @ left position.

Furthermore, it would be great, if it would be possible to assign them by choice to an Aux channel (potentiometer), so in flight tuning would be easier. 

Note: This was wrongly placed here: https://github.com/PX4/Firmware/issues/4067

DonLakeFlyer, 2016-03-23:
> Furthermore, it would be great, if it would be possible to assign them by choice to an Aux channel (potentiometer), so in flight tuning would be easier.

You can use RC to Param from parameter editor to assign changing any parameter to an RC channel. It's not very user friendly. But if you know how to do it it works. There is a separate issue alraedy in on creating a much better user friendly way to do all of this tweaking support.

closed 2016-03-23 by DonLakeFlyer

GervaisK, 2016-03-24:
Yes, it would be great if this could be integrated within a dropdown menue @ the tuning page.
Btw., it doesn´t work @ the moment, since the minimum / maximum values in the dialog window are´nt memorized. New issue to assign, or same reason as above ?

[image]

DonLakeFlyer, 2016-03-24:
Given the current api there isn't way to have min/max be remembered. The whole tuning thing needs a rework.