mavlink/qgroundcontrol #3071: Present command execution failures in a user-friendly way

LorenzMeier, 2016-03-24:
@DonLakeFlyer We do have logic to present results of an attempted command to the user here:
https://github.com/mavlink/qgroundcontrol/blob/39b220f8ebb86f41c95f93f1f7cdf4dac3b18535/src/uas/UAS.cc#L622-L642

Making the error message less cryptic and adding the command-name (user-facing name, not ENUM value) would help a lot. Once this works well the Firmware could then stop sending a text message in addition.

Note that this allows (at least in the case of PX4) also for command re-transmission, as you always get an ACK / NACK for a command if you request it.

DonLakeFlyer, 2016-03-24:
Is this already covered in #2975?

LorenzMeier, 2016-03-24:
No. I'm talking about user feedback. You talk about retransmission.

DonLakeFlyer, 2016-03-24:
If a command fails doesn't it come across as a critical error text message which then does a popup in QGC? If the text message is better from the firmware side isn't that good enough?

closed 2016-04-11 by DonLakeFlyer