mavlink/qgroundcontrol #3072: Look into auto-disconnect when link goes away

DonLakeFlyer, 2016-03-24:
Need to figure out how to deal with flaky UDP connections as well as OS differences on usb cable pull signalling across OS.

DonLakeFlyer, 2016-03-24:
Some discussion here as well: #3056

NaterGator, 2016-03-24:
@DonLakeFlyer on Windows QSerialPort::linkError is emitted with ResourceError: http://code.qt.io/cgit/qt/qtserialport.git/tree/src/serialport/qserialport_win.cpp?h=v5.5.1#n740

DonLakeFlyer, 2016-03-24:
If I remember correctly the issue is that the signalling differs between OS. OSX sends aboutToClose whereas Windows does not. It was a while ago I looked at this though. Maybe OSX signals that as well.

DonLakeFlyer, 2016-03-24:
Still not quite sure what to do with UDP though. WiFi connections come and go quite easily.

DonLakeFlyer, 2016-03-24:
But in thinking about it communication lost is just missing heartbeat, whereas these are device level signals. So I can distinguish between the two: plug pull, lost communication.

NaterGator, 2016-03-24:
Correct, semantically and in terms of user experience they are different things. Unreliable Wifi / network drops is just an indication of unreliable link. USB cord pulled (or OS level IO failure) is a more severe indication of a link fault that will definitely not resolve itself without user intervention. 

My OS X-fu is a little weak but I believe it returns ENXIO if the character device disappears, in which case the same error code will fire: http://code.qt.io/cgit/qt/qtserialport.git/tree/src/serialport/qserialport_unix.cpp?h=v5.5.1#n869

I'd say you're on the right track.

closed 2016-05-13 by DonLakeFlyer