mavlink/qgroundcontrol #3081: communication issue

jcooper21, 2016-03-25:
i was just setting up the AX10 frame in QGC and noticed it still says ready to fly once its connected shouldnt it say ready to drive?

also, i noticed there is not a reverse command in the drop down

DonLakeFlyer, 2016-03-26:
Rover is not really supported at this point. It may work to some extent, but has no testing against it and there is no cusrom plugin for it. So lots of things will be out of whack.

jcooper21, 2016-03-26:
Is it worth testing and giving feedback?

DonLakeFlyer, 2016-03-26:
Not really. There is too much missing. QGC should warn that Rover isn't supported. I'll fix that.

jcooper21, 2016-03-26:
That's too bad I really wanted to do a ground system

DonLakeFlyer, 2016-03-27:
I was looking into this some more. I think QGC may be fairly close to what it needs for Rover. I forgot that @pritamghanghas had added supported for Rover in the plugins. So testing and feedback would be good. Throw all your problems into this issue and I'll work through them.

jcooper21, 2016-03-27:
Unfortunately the board I use for testing just spontaneously lit on fire when I plugged it in so it'll be a couple weeks until I can test anything. Waiting on shipment now for some more test boards. I'd use Qualcomm flight but it doesn't have enough IO and the pin outs are different.

DonLakeFlyer, 2016-03-27:
Yikes!

DonLakeFlyer, 2016-06-23:
Rover should be mostly working now. You'll still see aircraft terminology bleed through. But things should be functional. We'll do more cleanup in 3.1 release.

DonLakeFlyer, 2016-06-23:
Should have said that it will be ready once this: https://github.com/mavlink/qgroundcontrol/pull/3626 gets into a daily build.

booo, 2021-10-30:
QGC supports rovers by now. If you encounter problems please create a specific ticket for the problem.

closed 2021-10-30 by booo