mavlink/qgroundcontrol #3082: APM plane always displaying armed

jcooper21, 2016-03-26:
When i connect to APM plane QGC always displays that it is armed even if its unarmed.

DonLakeFlyer, 2016-03-26:
I tried this with QGC master against ArduPilot master SITL and ArduPlane V3.5.1 on a board and in both cases it worked fine. Where are you seeing the armed indicator?

jcooper21, 2016-03-26:
At the bottom of the screen it always says its armed. I can neither disarm or arm and it verbally says the plane is armed when it first starts up. I am on APM 3.5.1. It always says its armed even when it's not.

jcooper21, 2016-03-26:
Could this be due to the arm switch?  I don't see a software arm parameter in APM.

DonLakeFlyer, 2016-03-26:
Can you give me a screenshot like this from the mavlink inspector widget?

[image]

jcooper21, 2016-03-26:
Sure. Give me 10 min

jcooper21, 2016-03-26:
[image]

DonLakeFlyer, 2016-03-26:
I need the HEARTBEAT entry expanded out so I can see values.

jcooper21, 2016-03-26:
[image]

DonLakeFlyer, 2016-03-26:
According to the value of base_mode the vehicle is armed. That is what the vehicle is reporting. So QGC is showing you the righ thing. Does APM Plane not have armed/disarmed concept?

jcooper21, 2016-03-26:
To arm I hold down the safety switch and hold it down again to disarm. The plane was disarmed in those screenshots. I would prefer to eliminate the switch and do a software arm on a tablet but not sure how with a plane.

jcooper21, 2016-03-26:
This also causes an issue for changing parameters as I have to go to mission planner since some parameters cannot be changed while armed.

DonLakeFlyer, 2016-03-26:
Hmm, seems to be some strange usage of the armed bits going on with ArduPlane. I have the safety switch pulled off my board. Let me put on back on and see what happens with my testing.

DonLakeFlyer, 2016-03-26:
I can't repro this with just my board. Let me get my fixed wing cut over to ArduPlane and I'll give that try.

jcooper21, 2016-03-26:
I'll disconnect my arm switch and try

DonLakeFlyer, 2016-03-26:
I tried it with a safety switch and it was still working correctly for me.

jcooper21, 2016-03-26:
I just disconnected my switch and tried again same result. I am using an HK32 pilot with rfd900 radios

jcooper21, 2016-03-26:
I tried a changing a bunch of arming parameters and couldn't get it to change. Mission planner does handle it correctly if that's any help.

DonLakeFlyer, 2016-03-26:
> Mission planner does handle it correctly if that's any help.

In what way?

jcooper21, 2016-03-26:
It displays disarmed and armed correctly and says safe when I hold the arm switch down.

DonLakeFlyer, 2016-03-26:
I was just running through a fixed wing setup on ArduPilot. It looks like a number of things are out of whack beyond just arming. I'm going to have to do a full run through of setup to find all the problems.

jcooper21, 2016-03-26:
In items in particular you found that I can check?

DonLakeFlyer, 2016-03-26:
RC Cal seems to be pretty out of whack. No need to check I need to debug it.

jcooper21, 2016-03-26:
Ya I think it is. It also likes to throw up a bunch of error messages when flying pre planned missions that were created in mission planner.  I just ignore it all at this stage.

DonLakeFlyer, 2016-03-26:
The RCMAP_* settings don't seem to work correctly which is odd since that's a firmware thing.

DonLakeFlyer, 2016-04-13:
Do you get these messages just connecting to the vehicle. Or do they come out when you are actively trying to arm/disarm the vehicle.

jcooper21, 2016-04-13:
That's connecting th vehicle.

DonLakeFlyer, 2016-04-13:
@jcooper21 Can you attach a zipped parameter set for your plane so I can repro your setup?

jcooper21, 2016-04-13:
Don please see parameters zip
[tuffwing parameters 3.zip](https://github.com/mavlink/qgroundcontrol/files/217636/tuffwing.parameters.3.zip)

jcooper21, 2016-04-17:
I found the issue its when the Arm_required parameter is set to 0 and the safety switch is only used to arm the plane QGC always says armed. If the value is set to 1 or 2 then QGC works correctly.

DonLakeFlyer, 2016-04-20:
If ARM_REQUIRED=0 then according to the parameter description the plane is armed immediately. So QGC is correctly stating that the plane is armed. Also with this set to 0 you can't disarm the plane either. So although a bit strange QGC is doing the right thing according to how you are parameters set.

jcooper21, 2016-04-20:
Interesting. With the safety switch mission planner says "safe" until the switch is pressed, so it's in a disarmed state which QGC does not.

closed 2016-04-20 by jcooper21

DonLakeFlyer, 2016-04-20:
Hmm. Still might be something out of whack here then. QGC doesn't really look for the "safe" state. Might need to be added.