mavlink/qgroundcontrol #3086: Problems with multiple vehicles on one link.

JMare, 2016-03-27:
When I have two vehicles on one link, both are APM:Plane, I encounter a problem in the following situation.

When I am viewing vehicle 2, if I depower vehicle 1 (Vehicle 2 still sending packets), and then switch to view vehicle 1, the vehicle selector will disappear and and I will not be able to switch back to vehicle 2, even though that one is still sending packets.

If I then press disconnect, it will disconnect both vehicles, and I will have to reconnect the link to regain connection to vehicle 2.

DonLakeFlyer, 2016-04-10:
For pull request:

The real fix is too involved as this point since we are trying to close down for 3.0. What this fix does is not disconnect if the link is being used by other Vehicles. The issues with this is that the inactive Vehicle will still hang around. But that's better than the current state of disconnecting the other active vehicle.

DonLakeFlyer, 2016-04-10:
Moving the real fix to 3.1

DonLakeFlyer, 2017-05-18:
I took another look through this. Of course I'm looking at this again right before a stable is going out. This is going to need to wait again. The temp fix is good enough for now.

DonLakeFlyer, 2018-01-05:
Now that the disconnect button is context sensitive to the vehicle selected I think this is working ok

closed 2018-01-05 by DonLakeFlyer