mavlink/qgroundcontrol #3111: Allow more zoom on the map

julianoes, 2016-03-29:
I think it would be better if the map could be zoomed closer, even if it involves scaling the available map tiles. The use case for me is to set a multicopter mission with waypoints close to each other.

As far as I remember, in previous versions it was possible to zoom in closer.

julianoes, 2018-10-01:
Is this hard to do? Could you give me pointers where I would have to dig? Thanks.

DonLakeFlyer, 2018-10-01:
I would assume it's quite involved. I means the map code needs to generate tiles on the fly (by blowing up lower zoom tiles) for the higher zoom levels which it doesn't have data for.

@dogmaphobic Would know more.

julianoes, 2018-10-01:
Ok, I understand, thanks.

booo, 2021-10-30:
Can you use a external tile server and a custom map url? As far as I can see there is an option for custom tile servers in QGC. Maybe it needs some documentation.

DonLakeFlyer, 2021-10-30:
This is "sort of" supported now. It will allow you to zoom up past the level where there are tiles and it blows up the existing tiles. I say "sort of" because you need to start lower than where tiles exist and then zoom up from there. If you start directly at the high zoom level it won't work. Better than nothing and likely as good as it's going to get.

closed 2021-10-30 by DonLakeFlyer