mavlink/qgroundcontrol #3112: Add auto-save of logs

julianoes, 2016-03-29:
I would vote to automatically save the mavlink logs instead of asking the user on shutdown every time.
Firstly, the "save log" dialog is annoying, and not wanted/needed most of the time. Secondly, I get so used to just ignore it and click it away that I do the same in the case when I actually want to keep the log.

dagar, 2016-03-29:
This would be a great option. Just auto save YYYY-MM-DD HH:MM:SS if armed?

DonLakeFlyer, 2016-04-01:
If folks are ok with having these just pile up over time this is doable.

julianoes, 2016-04-01:
I would be ok with it. Best would be a button in the settings: "delete all logs"

LorenzMeier, 2016-04-09:
@DonLakeFlyer Either keep all or rotate between 100-200 logs. Makes a lot of sense to me.

Antiheavy, 2017-01-31:
I'm late to the game, but I think this is a really good idea.  You might have multiple checkbox options:

- _Prompt to save Flight Data Log after each flight
- _Prompt to save Flight Data Log even if vehicle was not armed
- _Auto save Flight Data Log after each flight
- _Auto save Flight Data Log even if vehicle was not armed

closed 2017-03-11 by DonLakeFlyer

julianoes, 2017-03-11:
Awesome, thanks!