mavlink/qgroundcontrol #3120: QGC sometimes fails at saving parameters

tumbili, 2016-03-31:


DonLakeFlyer, 2016-04-01:
?

priseborough, 2016-04-04:
I have been unable to change parameters on the Snapdragon board using QGC with a UDP connection over wifi. I am able to refresh parameters, but changes do not take effect.

tumbili, 2016-04-04:
@DonLakeFlyer I'll do some testing in the lab today and will try to give you some more precise information.

tumbili, 2016-04-04:
I did a test with the Snapdragon. I changed the param "EKF2_MAG_TYPE" from automatic to heading mode. After I saved it a message came on the snapdragon terminal indicating that the param was changed. When I looked at the param in the terminal I confirmed that it has been changed and saved. However, after rebooting the board the parameter was at 0 (automatic heading fusion) again.
If I set the parameter directly in the terminal it works.
Also if I do it in QGC and then add a
```
param save
```
in the terminal at works. So I assume that the parameters are not actually saved.

DonLakeFlyer, 2016-04-04:
Here is what I did:
- I verified in the debugger that MAV_CMD_PREFLIGHT_STORAGE is being called in the right places after a parameter changes
- I verified that parameter change works across a reboot using a Pixhawk on Stable
- I verified that parameter change works across a reboot using a Pixhawk on Master

I couldn't seem to find the EKF2_MAG_TYPE parameter so I didn't use that specific param for testing. I don't have a Snapdragon so I can't test directly with it. But to me, everything points to a firmware problem, not a QGC problem.

DonLakeFlyer, 2016-04-04:
Can you verify if this is a Snapdragon only problem?

julianoes, 2016-04-05:
I can reproduce the issue on Snapdragon but not on Pixhawk, so I think it's specific to Snapdragon. I'll do more digging.

julianoes, 2016-04-05:
This was related to the shared memory stuff on Snapdragon. The QGC side seems right.
Fixed in https://github.com/PX4/Firmware/pull/4168.

closed 2016-04-05 by julianoes

priseborough, 2016-04-06:
I still have problems with getting param changes to stick on my snapdragon

julianoes, 2016-04-06:
Which branch commit did you test?

priseborough, 2016-04-06:
I just replicated the fault using the following commit from catchup-1

20a7acee6798a34de4fac9cc1673250348b61dfa

julianoes, 2016-04-07:
@priseborough this was because of the missing file system sync, correct?

priseborough, 2016-04-07:
I think so - since I have been starting an ssh term over the wifi and manually running the sync command after changing parameters over QGC, the fault has not not re-occurred.