mavlink/qgroundcontrol #3125: cannot install qt5.5+ on NVIDIA Jetson tk1 for qgroundcontrol

jeansabu1, 2016-04-01:
Hi. I tried installing qgroundcontrol on jetson tk1, and encountered a problem with qt version.
Qgroundcontrol needs qt5.5+, but I could install only upto qt5.2 on jetson tk1.
The reason I wanted to run qgroundcontrol was to disable ARMING_CHECK to resolve the error message: 
      APM: PreArm: RC not calibrated
I could disable the ARMING_CHECK on my laptop. But, on jetson tk1 I could not run qgroundcontrol because of the aforementioned qt version problem.

I would appreciate any help.

Sincerely yours,
Jin

DonLakeFlyer, 2016-04-02:
Not much you can do here. QGC needs that version of Qt. If your hardware can't support it, then the hardware can't support QGC.

closed 2016-04-02 by DonLakeFlyer

jeansabu1, 2016-04-06:
Is there any way I can disable ARMING_CHECK by using command-line over mavlink?

NaterGator, 2016-04-06:
@DonLakeFlyer @jeansabu1,  the TX1 is definitely capable of supporting Qt 5.5.1. Unfortunately I don't know of a PPA offering armhf binaries for the L4T 23.2 release so it is up to you to cross-compile (or natively compile) yourself.

DonLakeFlyer, 2016-04-06:
> Is there any way I can disable ARMING_CHECK by using command-line over mavlink?

I think there is a parameter you can set to turn it off