mavlink/qgroundcontrol #3127: clear mavlink log

dagar, 2016-04-01:
Can we get a button to clear the mavlink messages?

[image]

NaterGator, 2016-04-20:
@DonLakeFlyer I find the whole singleton UASMessageHandler architecture a bit contrived. Do you mind if I pick this up and move UASMessages into a model stored on Vehicle? We can then switch this message console to a model-view based architecture that makes more sense.

dogmaphobic, 2016-04-20:
Yes,  I agree. That code is some ancient left over from before any of the QtQuick UI was ever in place.

dogmaphobic, 2016-05-06:
> Do you mind if I pick this up and move UASMessages into a model stored on Vehicle? We can then switch this message console to a model-view based architecture that makes more sense.

It makes it weird because it's still used by a Qt Widget *Widget*. If we get rid of that, it can be made a lot simpler. I had to hack some formatting (#3344) so it could be used in both places.

dagar, 2016-05-06:
Thanks @dogmaphobic!

closed 2016-05-06 by dagar

DonLakeFlyer, 2016-05-07:
Toss the qt widget support