mavlink/qgroundcontrol #3128: "Vehicle did not respond to request for parameters" on one Android device but not the other

dzett, 2016-04-01:
Hi there,

I am running the current beta of QGC (2.9.4-280-g38090bf) on two Android devices: A Nexus 5 phone and an older Tablet connected via Wifi to a Pixracer with the current beta build of PX4. 

On the Phone, I get periodic error messages "Vehicle did not respond to request for parameters, retrying" although the map/instrument view is updated fine. The settings/parameters page is not accessible.

[image]
[image]

On the Tablet, the issue does not happen (exactly same version). Daily build of the QGC Linux version on a PC connected via USB also is happy.
I have: Deleted all App data and re-installed the apps on both devices but it does not change.

The issue also happened with earlier master builds of PX4 and also with earlier beta versions of QGC, but there was some QGC version in between in the last three weeks where it did work correctly also on the phone.

Any clue? Is there a way I can support analysis by providing a log of some kind?

Thanks,
Daniel

DonLakeFlyer, 2016-04-02:
Ok, so it is connected and getting heartbeats. But the vehicle is not responding to parameter requests. This is most likely an issue on the firmware side. 

@LorenzMeier Can you determine how the vechile is getting/responding to the param requests from teh sd logs?

dzett, 2016-04-02:
Nevermind, I found out the cause:

It seems to be related to how Android 6 handles Wifi connections without internet connectivity.

Behavior is reproducible: After deleting the Pixracer Wifi network in Android and re-adding it, Android will show a "Wifi-Question-Mark" icon / message informing about missing internet connectivity. 
_So far, I would so far just swipe it away to disappear._ - not a good idea.
When opening it, a dialog is shown that asks (sorry, screenshot German only):

[image]

"This network does not have internet access. Do you want to keep the connection?". If you answer "Yes" and tick "Don't as again", QGC immediately shows the progress bar and has data.
I guess swiping the dialog away keeps it in some odd "neither-yes-nor-no" state, where it doesn't disconnect (and accepts incoming UDP traffic?) but doesn't allow outbound connections.

This is on an Android 6.0.1 device. My older tablet (Android 4.1.1) doesn't have a similar dialog and just connects straight through (it shows the Wifi "grey" as there is not internet, but not questions or nagging...).

So, this is "solved" for me and not really a QGC issue. As briefly mentioned on gitter, maybe it would be possible to add a check to QGC (heartbeat ok but parameter request fails) and show a hint like "please re-add the FC Wifi network and accept it for connections" (not an easy one to word though). What do you think?

DonLakeFlyer, 2016-04-04:
Maybe this: http://developer.android.com/reference/android/net/wifi/WifiManager.html#enableNetwork%28int,%20boolean%29

DonLakeFlyer, 2016-04-10:
@dogmaphobic Does any of this make sense to you?

DonLakeFlyer, 2016-05-13:
Need to research this some more to figure out if there is some setting we need to change.

DonLakeFlyer, 2016-07-01:
@dogmaphobic Do we need to fix some wifi without internet thing for 3.0? Or move to 3.1? Or is there even a problem here?

dogmaphobic, 2016-07-01:
I assume there is some (Java) interface to tell Android to buzz off and let the app use WiFi even though it's not connected to the "Internet". I won't have the time to deal with Android before 3.1.

DonLakeFlyer, 2016-07-01:
Moving to 3.1

dogmaphobic, 2016-12-31:
Could someone test this once #4380 makes it in?

DonLakeFlyer, 2017-05-26:
If this is still broken in 3.2 we live with it. Moving to fixing in first phase of 3.3

DonLakeFlyer, 2018-01-04:
Hoping it's fixed now

closed 2018-01-04 by DonLakeFlyer